tinygo-org/tinygo #575: Support fuction reflect for wasm

TerryGuo, 2019-09-20:
Hi there,

We are trying to implement function reflect for wasm target. It seems wasm has some limitations stopping us to do so. Do you guys have any thoughts on this topic? Or is this an impossible mission? 

BR,
Terry

By fucntion reflect, I mean code like below:
```C++
package main

import "reflect"

func foo(x int) int {
	return x
}

func bar(f reflect.Value) {
	in := []reflect.Value{reflect.ValueOf(12345)}
	out := f.Call(in)
	println(out[0].Int())
}

func main() {
	f := reflect.ValueOf(foo)
	bar(f)
}
```

jadr2ddude, 2019-09-20:
We do not currently support using reflection for function calls. This should eventually be supported, but this is one of the more complex pieces of functionality in reflect and nobody has quite gotten around to it. It is possible that this could be implemented by having the compiler generate wrappers.

TerryGuo, 2019-09-21:
Thanks Jaden. We will give your suggestion a try.

kwunyeung, 2019-09-24:
I got the following errors while compiling. Does it relate to this issue?

```
# encoding/asn1
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/marshal.go:537:47: v.Type().NumMethod undefined (type reflect.Type has no field or method NumMethod)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/marshal.go:549:14: DeepEqual not declared by package reflect
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/marshal.go:558:14: DeepEqual not declared by package reflect
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/marshal.go:479:93: t.Field(startingField).Tag.Get undefined (type string has no field or method Get)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/marshal.go:483:103: t.Field(i + startingField).Tag.Get undefined (type string has no field or method Get)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/common.go:174:26: t.Name undefined (type reflect.Type has no field or method Name)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/asn1.go:658:80: ifaceType.NumMethod undefined (type reflect.Type has no field or method NumMethod)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/asn1.go:810:115: fieldType.Name undefined (type reflect.Type has no field or method Name)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/asn1.go:831:12: Copy not declared by package reflect
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/asn1.go:919:104: field.Tag.Get undefined (type string has no field or method Get)
../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/asn1.go:932:12: Copy not declared by package reflect
```

jadr2ddude, 2019-09-24:
No that is a seperate issue - reflect is incomplete and anything relying on complex functionality is unlikely to work yet

northka, 2020-03-11:
I met the same  problem
hope to fixed the problem faster

northka, 2020-03-11:
> I got the following errors while compiling. Does it relate to this issue?
>
> ```
> # encoding/asn1
> ../../../../usr/local/Cellar/go@1.12/1.12.9/libexec/src/encoding/asn1/marshal.go:537:47: [...]

I met the same problem
hope to fixed the problem faster

aykevl, 2020-03-11:
@northka I've started work on that in https://github.com/tinygo-org/tinygo/pull/907.

deadprogram, 2021-04-22:
Since the issue from #907 was solved differently can this issue now be closed?

dgryski, 2023-03-23:
WASI (and tinygo in general) now supports reflection.

aykevl, 2023-05-03:
@dgryski I don't think this is actually fixed: the bug is about `.Call()` which is not yet supported.

dgryski, 2023-05-03:
Oh good point.  I was responding to the second issue about `encoding/asn1`.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

Sexual, 2023-10-20:
Any updated plans to support `reflect.FuncOf`? This is something we really need.

T-J-A, 2024-07-16:
Is there any movement on this? This is something I'd really love to use.

closed 2026-11-15 by deadprogram