tinygo-org/tinygo #586: Stack OverFlow Error

SuryaprakashBS, 2019-09-23:
runtime: goroutine stack exceeds 1000000000-byte limit
fatal error: stack overflow

runtime stack:
runtime.throw(0x34b04a2, 0xe)
	/usr/local/go/src/runtime/panic.go:617 +0x72
runtime.newstack()
	/usr/local/go/src/runtime/stack.go:1041 +0x6f0
runtime.morestack()
	/usr/local/go/src/runtime/asm_amd64.s:429 +0x8f

goroutine 1 [running]:
runtime.rawstringtmp(0xc03c0004b8, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/runtime/string.go:118 +0xa0 fp=0xc03c000348 sp=0xc03c000340 pc=0x8e6280
runtime.concatstrings(0xc03c0004b8, 0xc03c000428, 0x3, 0x3, 0x0, 0x0)
	/usr/local/go/src/runtime/string.go:49 +0xae fp=0xc03c0003e0 sp=0xc03c000348 pc=0x8e5cde
runtime.concatstring3(0xc03c0004b8, 0x34acdc5, 0x7, 0x34ab895, 0x1, 0xc00092f8d4, 0xa, 0x0, 0x0)
	/usr/local/go/src/runtime/string.go:62 +0x47 fp=0xc03c000420 sp=0xc03c0003e0 pc=0x8e5fc7
github.com/tinygo-org/tinygo/compiler.(*Compiler).getLLVMType(0xc000178000, 0x352e3c0, 0xc002a3f6b0, 0x0)
	/home/circleci/project/compiler/compiler.go:500 +0x1021 fp=0xc03c000538 sp=0xc03c000420 pc=0xb7ed91
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc002a3f6b0, 0xa)
	/home/circleci/project/compiler/compiler.go:580 +0x5d fp=0xc03c000b98 sp=0xc03c000538 pc=0xb7f05d
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e340, 0xc0000e0af0, 0x604da60)
	/home/circleci/project/compiler/compiler.go:647 +0x1801 fp=0xc03c0011f8 sp=0xc03c000b98 pc=0xb80801
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc0000944b0, 0x4)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c001858 sp=0xc03c0011f8 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c001eb8 sp=0xc03c001858 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c002518 sp=0xc03c001eb8 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc000b71380, 0x60b1a30)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c002b78 sp=0xc03c002518 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc0052aa5c0, 0xc01cba1980)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c0031d8 sp=0xc03c002b78 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e480, 0xc000b712e0, 0xc)
	/home/circleci/project/compiler/compiler.go:697 +0xba1 fp=0xc03c003838 sp=0xc03c0031d8 pc=0xb7fba1
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c003e98 sp=0xc03c003838 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c0044f8 sp=0xc03c003e98 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc000b71380, 0x60b1a30)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c004b58 sp=0xc03c0044f8 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc0052aa400, 0xc01cba1800)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c0051b8 sp=0xc03c004b58 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e480, 0xc000b712e0, 0xc)
	/home/circleci/project/compiler/compiler.go:697 +0xba1 fp=0xc03c005818 sp=0xc03c0051b8 pc=0xb7fba1
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c005e78 sp=0xc03c005818 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c0064d8 sp=0xc03c005e78 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc000b71380, 0x60b1a30)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c006b38 sp=0xc03c0064d8 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc0052aa240, 0xc01cba1680)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c007198 sp=0xc03c006b38 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e480, 0xc000b712e0, 0xc)
	/home/circleci/project/compiler/compiler.go:697 +0xba1 fp=0xc03c0077f8 sp=0xc03c007198 pc=0xb7fba1
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c007e58 sp=0xc03c0077f8 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c0084b8 sp=0xc03c007e58 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc000b71380, 0x60b1a30)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c008b18 sp=0xc03c0084b8 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc0052aa080, 0xc01cba14c0)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c009178 sp=0xc03c008b18 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e480, 0xc000b712e0, 0xc)
	/home/circleci/project/compiler/compiler.go:697 +0xba1 fp=0xc03c0097d8 sp=0xc03c009178 pc=0xb7fba1
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c009e38 sp=0xc03c0097d8 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c00a498 sp=0xc03c009e38 pc=0xb81147
[... 126 lines cut ...]
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c023c98 sp=0xc03c023638 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c0242f8 sp=0xc03c023c98 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc000b71380, 0x60b1a30)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c024958 sp=0xc03c0242f8 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc0052a8800, 0xc01cb91f00)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c024fb8 sp=0xc03c024958 pc=0xb7f5d0
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e480, 0xc000b712e0, 0xc)
	/home/circleci/project/compiler/compiler.go:697 +0xba1 fp=0xc03c025618 sp=0xc03c024fb8 pc=0xb7fba1
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e4c0, 0xc003212db0, 0x60b19f0)
	/home/circleci/project/compiler/compiler.go:730 +0x349 fp=0xc03c025c78 sp=0xc03c025618 pc=0xb7f349
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e3c0, 0xc003212d80, 0x60b1a10)
	/home/circleci/project/compiler/compiler.go:652 +0x2147 fp=0xc03c0262d8 sp=0xc03c025c78 pc=0xb81147
github.com/tinygo-org/tinygo/compiler.(*Compiler).getDIType(0xc000178000, 0x352e400, 0xc000b71380, 0x60b1a30)
	/home/circleci/project/compiler/compiler.go:657 +0x5d0 fp=0xc03c026938 sp=0xc03c0262d8 pc=0xb7f5d0
...additional frames elided...

goroutine 19 [syscall, 1 minutes]:
os/signal.signal_recv(0x0)
	/usr/local/go/src/runtime/sigqueue.go:139 +0x9c
os/signal.loop()
	/usr/local/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
	/usr/local/go/src/os/signal/signal_unix.go:29 +0x41

jadr2ddude, 2019-09-23:
Can you please share the code that triggered this error?

SuryaprakashBS, 2019-09-23:
Sorry, I can't share the code. I will try to reproduce this this error using test code and send you later.

aykevl, 2019-09-23:
I think I know where this is coming from, but a small sample would be very helpful.

I think you can work around this with `-no-debug`.

SuryaprakashBS, 2019-09-27:
test code:
package main

import (
	"fmt"
)

type itemType string

const (
	BLOCK itemType = "block"
)

func test() map[itemType]string {
	testMap := map[itemType]string{}
	testMap[BLOCK] = "test"
	return testMap
}

func main() {
	fmt.Println(test()[BLOCK])
}

output:
bug.go:21:23: only strings, bools, ints, pointers or structs of bools/ints are supported as map keys, but got: bug.go.itemType

aykevl, 2019-09-27:
That error (`only strings, bools, ints, pointers or structs of bools/ints are supported as map keys, but got: bug.go.itemType`) is a different bug, unrelated to the stack overflow.

jadr2ddude, 2019-10-07:
I believe that this issue is the cause behind the problem of #598. So, #598 is blocked on this.

jadr2ddude, 2019-10-17:
This should be fixed in v0.9

closed 2019-10-17 by jadr2ddude

SuryaprakashBS, 2019-10-21:
I want this fixed version .Where can I get this? dev branch?

jadr2ddude, 2019-10-21:
https://github.com/tinygo-org/tinygo/releases/tag/v0.9.0

PR #619 (2019-10-13): compiler: support recursive types
Previously, the cycle was broken by inserting an `unsafe.Pointer` type in some places. This is of course incorrect, and makes debugging harder. However, LLVM provides a way to make temporary nodes that are later replaced, exactly for this purpose.

This commit uses those temporary metadata nodes to allow such recursive types.

Fixes #586.