tinygo-org/tinygo #594: ROM segments are non-contiguous with LLVM9

QuLogic, 2019-09-26:
When building with LLVM 9, all of the smoke tests except wasm and avr fail with this error:
```
tinygo build -size short -o test.hex -target=pca10040            examples/blinky1
   code    data     bss |   flash     ram
   5340      60     196 |    5400     256
error: ROM segments are non-contiguous: /tmp/tinygo703550982/main
```
There's not much more information, so I don't know if LLVM is stricter or tinygo is outputting something wrong now.

aykevl, 2019-09-26:
Thank you for the report! This is a problem in TinyGo and should be fixed. It looks like the lld linker was changed significantly.

aykevl, 2019-11-12:
Turns out lld decided to fill the `.stack` section with some arbitrary pattern, and thereby marking it as `PROGBITS` instead of `NOBITS`. This should be fixed in #715.

aykevl, 2019-11-17:
The fix for this has been merged in the dev branch.

PR #715 (2019-11-12): targets: explicitly mark the stack as NOLOAD
This prevents it from being of type PROGBITS in lld 9, it should always be NOBITS. This should fix the following error in lld 9:

    ROM segments are non-contiguous

Note: I haven't yet tested this.

fixes #594
closed 2019-11-17 by aykevl