tinygo-org/tinygo #595: Builtins' symbol index is no longer optional

QuLogic, 2019-09-26:
In `compileBuiltins`, there is a comment:
https://github.com/tinygo-org/tinygo/blob/cc313798f381532e4874a7166fa3c216234c005d/builtins.go#L248-L250
But with LLVM9, this appears no longer true. Running smoke tests fail to link:
```
tinygo build -size short -o test.hex -target=pca10040            examples/blinky1
ld.lld: error: .../tinygo/pkg/armv7em-none-eabi/compiler-rt.a: archive has no index; run ranlib to add one
error: failed to link /tmp/tinygo976334510/main: exit status 1
```

I could not find any symbol index writer in blakesmith/ar, so for now I just run `ranlib` before installing the `compiler-rt.a` files.

aykevl, 2019-09-26:
Oops, this definitely should be fixed. Although you can work around it by running `ranlib` and that should be sufficient for a distro package, I would really like to avoid requiring such an external tool (that's why TinyGo itself started generating these files in the first place). I'll see what I can do.

aykevl, 2019-11-12:
This is the relevant LLVM change: https://reviews.llvm.org/D63781

MaskRay, 2019-11-12:
This change is intentional. GNU linkers do not accept archives without a symbol table. lld used to be loose before https://reviews.llvm.org/D63781

```
% ld.bfd a.o a.a
ld.bfd: a.a: error adding symbols: archive has no index; run ranlib to add one
% gold a.o a.a
gold: error: a.a: no archive symbol table (run ranlib)
```

sbc100, 2019-11-12:
IIUC, before this change, lld would silently ignore libraries without symbol tables.  So I'm surprised this ever worked here (or perhaps there was a different change involved?).

aykevl, 2019-11-12:
In LLVM 8, lld was definitely able to read archives without a symbol table and we've relied on that behavior. For example, it was possible to use floating point operations on a Cortex-M0 (that has no FPU) after linking with compiler-rt, and this library definitely did not have a symbol table. This behavior breaks now we're trying to upgrade to LLVM 9.

I tried adding an empty symbol table for lld 9 (first file set to name `/` and containing 4 null bytes) but lld reports lots of error of undefined symbols. So it looks like either lld is now unable to read the individual members or (perhaps more likely) doesn't even attempt to read them when it sees a symbol table.

I'm not arguing for reverting the behavior. Adding a symbol table seems like the proper way to go, and probably allows lld to be faster (even though that really doesn't matter for this use case: the library is 388kB). I'm working on a patch now to do exactly that.

aykevl, 2019-11-12:
See #717 for a fix.

aykevl, 2019-11-17:
This has been implemented in the dev branch.

closed 2019-11-17 by aykevl

MaskRay, 2022-01-14:
After 2 years, I think perhaps an archive without index should be supported. Created https://reviews.llvm.org/D117284 ([ELF] Allow non-bitcode archive with an empty index).

aykevl, 2022-01-21:
Hmm, but I think it would be best to keep the index for now? Removing the code to create the index would be nice though.