tinygo-org/tinygo #615: flash: perform file copy for UF2 bootloader platforms

deadprogram, 2019-10-11:
For all platforms that support the UF2 bootloader, the flashing of new code takes place by performing a file copy to the drive mounted for the device to be flashed.

For example, on the Adafruit Circuit Playground Express, the flashing takes place by compiling the code then copying it to `/media/[USERNAME]/[NAME OF THE BOARD]/flash.uf2` when running on a Linux-based OS.

I propose that we look to the output file type here https://github.com/tinygo-org/tinygo/blob/master/main.go#L410 then perform the file copy by locating the expected file path based on the info in the `flash` key, in similar flash to how this script does so: https://github.com/microsoft/uf2/blob/master/utils/uf2conv.py#L173

For example, this key `/media/*/ITSYM4BOOT/{uf2}` would tell `tinygo flash` to compile to UF2, then perform the file copy by finding the needed directory using https://golang.org/pkg/path/filepath/#Glob

This would not change the normal behavior of the `tinygo build` command for UF2 output, just `tinygo flash`.

deadprogram, 2019-10-11:
PR #617 addresses this.

aykevl, 2019-10-12:
I think this can be done in a somewhat cleaner way with these two modifications:

1. By adding a `flash-method` (or similar) key, that selects which method to use for flashing. Initially `command` and `msd` (mass-storage device), but could later be extended with other methods.
2. By adding a `flash-msd-volume-name` key, which sets the name of the volume. This way you could have multiple boards connected to a computer and only flash the one you intend to flash. Also, it provides some protection against accidentally copying to a regular USB drive.

In the future, we might have a `flash-method` of `openocd`, and you'd just set the chip name and programmer name. The advantage of that is that the GDB command (for on-chip debugging) could be automatically inferred and that board inheritance works a bit cleaner. Also, it might be possible to add a command line flag to select a programmer, something that could be very useful for boards that require an independent programmer (and could therefore be programmed with a variety of programmers). For example, you could do `tinygo flash -programmer=stlinkv2` or `tinygo flash -programmer=bmp` depending on what programmer you have connected that moment.

deadprogram, 2019-10-13:
OK I have updated #617 based on this feedback. I've also modified a couple of other targets to support similar MSD flashing using hex files where supported, such as the Microbit and the HiFive1.

deadprogram, 2019-10-13:
This has been merged into `dev` so now closing.

closed 2019-10-13 by deadprogram