tinygo-org/tinygo #618: Stack overflow compilation error, SIGILL instead

gsora, 2019-10-11:
```

jadr2ddude, 2019-10-11:
The first problem would be #598 I think

gsora, 2019-10-11:
I tried building with that fork, it wouldn't compile with or without `-no-debug`

jadr2ddude, 2019-10-11:
Oops, meant #586

jadr2ddude, 2019-10-11:
The second issue is something else

gsora, 2019-10-11:
With `-no-debug` the binary simply crashes with `SIGILL`.

aykevl, 2019-10-12:
Yes, this is a duplicate of #586, so closing (but noting that this is a high priority issue).

aykevl, 2019-10-12:
The error with `-no-debug` is probably a different bug.

jadr2ddude, 2019-10-12:
We may want to reopen this and instead retitled it to focus on the sigil instead

aykevl, 2019-10-12:
Sure, that's also an option.
@gsora feel free to update this issue with the other bug you discovered, and reopen it.

gsora, 2019-10-15:
There isn't much to add except renaming the issue.

The sample program I provided does crash with SIGILL on a Linux/ARM target, as well as Linux/x86_64.

Any given permutation of `-gc` doesn't change anything.

Since this program only compiles with `-no-debug` there isn't much to work with, but gdb seem to understand something anyway:

```
Starting program: /home/gsora/hellosphere-go/hellosphere-go

Program received signal SIGILL, Illegal instruction.
0x000000000040d75a in (*sync.Pool).Get ()
```

gsora, 2019-10-15:
Maybe something's wrong with the code path taken by `fastjson.GetString()`, because the following program works fine:

```go
package main

import (
	"fmt"
	"github.com/valyala/fastjson"
	"log"
)

func main() {
	s := `{"string":"yes!"}`
	fmt.Println(fastjson.Validate(s))
	var p fastjson.Parser
	v, err := p.Parse(`{
		"str": "bar",
		"int": 123,
		"float": 1.23,
		"bool": true,
		"arr": [1, "foo", {}]
	}`)

	if err != nil {
		log.Fatal(err)
	}

	fmt.Printf("foo=%s\n", v.GetStringBytes("str"))
	fmt.Printf("int=%d\n", v.GetInt("int"))
	fmt.Printf("float=%f\n", v.GetFloat64("float"))
	fmt.Printf("bool=%v\n", v.GetBool("bool"))
	fmt.Printf("arr.1=%s\n", v.GetStringBytes("arr", "1"))
}

gsora, 2019-10-15:
`fastjson.GetString()` access an instance of `fastjson.ParserPool` when the function is first called, and the SIGILL definitely comes from there: https://github.com/valyala/fastjson/blob/master/pool.go#L8

jadr2ddude, 2019-10-16:
I believe that the SIGILL is caused by #623, and it should have been a nil panic instead. That said, I have not yet figured out why we end up with a nil function.

jadr2ddude, 2019-10-16:
Actually, the nil function was in fact legal here. There appears to be an inconsistency between our current implementation of `sync.Pool` and that which is in the standard go implementation.

gsora, 2019-10-16:
With #624, the compiling process fails:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x6f81de]

goroutine 1 [running]:
golang.org/x/tools/go/ssa.(*Type).Type(...)
        /home/gsora/go/src/golang.org/x/tools/go/ssa/ssa.go:1465
github.com/tinygo-org/tinygo/compiler.(*Compiler).getRuntimeType(...)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:439
github.com/tinygo-org/tinygo/compiler.(*Compiler).getLLVMRuntimeType(0xc000170000, 0x84cb2a, 0xb, 0x77d0a0)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:446 +0xae
github.com/tinygo-org/tinygo/compiler.(*Compiler).makeStructTypeFields(0xc000170000, 0xc000e0cc90, 0xc00111a0c0)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:96 +0x55
github.com/tinygo-org/tinygo/compiler.(*Compiler).getTypeCode(0xc000170000, 0x8cb980, 0xc000e0cc90, 0x2b)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:72 +0x54a
github.com/tinygo-org/tinygo/compiler.(*Compiler).getTypeCode(0xc000170000, 0x8cb880, 0xc000e0cc60, 0x33)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:60 +0x4d7
github.com/tinygo-org/tinygo/compiler.(*Compiler).getTypeCode(0xc000170000, 0x8cb8c0, 0xc000a5e1c0, 0x1)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:64 +0x38d
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseMakeInterface(0xc000170000, 0x7f7bac543eb8, 0x8cb8c0, 0xc000a5e1c0, 0x0, 0x0)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/interface.go:27 +0xb3
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseExpr(0xc000170000, 0xc00530f710, 0x8daf00, 0xc0029560a0, 0xc0029560a0, 0x1, 0x0)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:1626 +0x2415
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseInstr(0xc000170000, 0xc00530f710, 0x8daea0, 0xc0029560a0)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:1013 +0x1691
github.com/tinygo-org/tinygo/compiler.(*Compiler).parseFunc(0xc000170000, 0xc00530f710)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:987 +0xe44
github.com/tinygo-org/tinygo/compiler.(*Compiler).Compile(0xc000170000, 0x8494d5, 0x1, 0xd, 0x0, 0x0)
        /home/gsora/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:337 +0xa3d
main.Compile(0x8494d5, 0x1, 0x7ffffa0a0a3e, 0xe, 0xc000086600, 0xc004149ee0, 0xc004149d48, 0x0, 0x0)
        /home/gsora/go/src/github.com/jadr2ddude/tinygo/main.go:125 +0xa94
main.Build(0x8494d5, 0x1, 0x7ffffa0a0a3e, 0xe, 0x0, 0x0, 0xc0000b3ee0, 0xc000124250, 0x7f7be490cb18)
        /home/gsora/go/src/github.com/jadr2ddude/tinygo/main.go:337 +0xe2
main.main()
        /home/gsora/go/src/github.com/jadr2ddude/tinygo/main.go:696 +0xefd
```

jadr2ddude, 2019-10-16:
I am sorry, but I cannot replicate that error.

You also do not appear to have a correctly built copy of tinygo - and the stack trace includes two different copies of tinygo. The following series of commands should work if run inside of `/home/gsora/go/src/github.com/tinygo-org/tinygo`:
```
git fetch origin pull/624/head:syncpoolnilnew
git checkout syncpoolnilnew
go install -v
```

On my PR, your original piece of code compiled and ran:
```
[jadenw@jaden-desktop tinygo]$ build/tinygo run /tmp/fastjson.go 
yes!
```

gsora, 2019-10-16:
You're absolutely right, my builds were surely broken.

The SIGILL has been fixed in #624.

closed 2019-10-16 by gsora