tinygo-org/tinygo #627: WASM: panic when trying to compile a file included the init func contains fmt.Sprint()

hmgle, 2019-10-16:
Env:

```
$ tinygo version
tinygo version 0.8.0 linux/amd64
```

main.go:

```go
package main

import "fmt"

var Foo string

func main() {
	fmt.Println(Foo)
}

func init() {
	Foo = fmt.Sprint("foo")
}
```

Command:

```
tinygo build -o main.wasm -target wasm main.go
```

Result:

```
panic: todo: store

goroutine 1 [running]:
github.com/tinygo-org/tinygo/interp.(*Eval).hasLocalSideEffects(0xc0017ebce0, 0xc000e920a8, 0x7f183020e7a8, 0xc001098c00)
        /home/circleci/project/interp/scan.go:179 +0x1d2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f183002b248, 0x7f183002b248)
        /home/circleci/project/interp/scan.go:106 +0x555
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f183002bfb8, 0x7f183002bfb8)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f1830032ef8, 0x7f1830032ef8)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f18300320e8, 0x7f18300320e8)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f1830032e08, 0x7f1830032e08)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f1830030868, 0x7f1830030868)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f183002f758, 0x7f183002f758)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*Eval).hasSideEffects(0xc0017ebce0, 0x7f1830030688, 0x1)
        /home/circleci/project/interp/scan.go:100 +0x4f2
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc000e932b8, 0x7f1830f89790, 0x0, 0x0, 0x0, 0xc031326638, 0x10, 0x10, 0xc00239dd60, 0xa704c1, ...)
        /home/circleci/project/interp/frame.go:426 +0x4d5f
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0017ebce0, 0x7f1830030598, 0xc000e933f8, 0x2, 0x2, 0xc0023a3e80, 0x7, 0x0, 0x0, 0x0, ...)
        /home/circleci/project/interp/interp.go:104 +0x208
github.com/tinygo-org/tinygo/interp.(*Eval).Function(...)
        /home/circleci/project/interp/interp.go:87
github.com/tinygo-org/tinygo/interp.Run(0x66817c0, 0x667fcc0, 0x0, 0x0, 0x0)
        /home/circleci/project/interp/interp.go:74 +0x681
main.Compile(0x7fffdb22ceec, 0x7, 0x7fffdb22ced5, 0x9, 0xc00011a4e0, 0xc000e93ed0, 0xc000e93d28, 0x0, 0x0)
        /home/circleci/project/main.go:148 +0xd02
main.Build(0x7fffdb22ceec, 0x7, 0x7fffdb22ced5, 0x9, 0x7fffdb22cee7, 0x4, 0xc00008bed0, 0xc0000ec310, 0xffffffffffffffff)
        /home/circleci/project/main.go:339 +0xe2
main.main()
        /home/circleci/project/main.go:702 +0xfd7
```

main2.go:

```go
package main

import "fmt"

var Foo = fmt.Sprint("foo")

func main() {
	fmt.Println(Foo)
}
```

Command:

```
tinygo build -o main2.wasm -target wasm main2.go
```

Result:

```
panic: interp: GEP on a constant

goroutine 1 [running]:
github.com/tinygo-org/tinygo/interp.(*LocalValue).GetElementPtr(0xc000dbb820, 0xc000dc0fe8, 0x2, 0x2, 0x353f000, 0xc000dbb810)
        /home/circleci/project/interp/values.go:107 +0x288
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc00262d2b8, 0x7f1cbcf89f00, 0x7f1cbcf89b80, 0x0, 0x0, 0x0, 0x0, 0xc0029d50e8, 0x1, 0x1, ...)
        /home/circleci/project/interp/frame.go:132 +0x1e86
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc002087890, 0x7f1cbc030598, 0xc00262d3f8, 0x2, 0x2, 0xc000dc0e50, 0x7, 0x0, 0x0, 0x0, ...)
        /home/circleci/project/interp/interp.go:104 +0x208
github.com/tinygo-org/tinygo/interp.(*Eval).Function(...)
        /home/circleci/project/interp/interp.go:87
github.com/tinygo-org/tinygo/interp.Run(0x4a28680, 0x4a26b80, 0x0, 0x0, 0x0)
        /home/circleci/project/interp/interp.go:74 +0x681
main.Compile(0x7ffe5b25ceec, 0x7, 0x7ffe5b25ced5, 0x9, 0xc0001224e0, 0xc00262ded0, 0xc00262dd28, 0x0, 0x0)
        /home/circleci/project/main.go:148 +0xd02
main.Build(0x7ffe5b25ceec, 0x7, 0x7ffe5b25ced5, 0x9, 0x7ffe5b25cee7, 0x4, 0xc000099ed0, 0xc0000f2290, 0xffffffffffffffff)
        /home/circleci/project/main.go:339 +0xe2
main.main()
        /home/circleci/project/main.go:702 +0xfd7
```

main3.go:

```go
package main

import "fmt"

var Foo string

func main() {
	Foo = fmt.Sprint("foo")
	fmt.Println(Foo)
}
```

Command:

```
tinygo build -o main3.wasm -target wasm main3.go
```

Result: OK.

deadprogram, 2019-10-18:
Is this still an issue with the latest 0.9.0 release?

jadr2ddude, 2019-10-18:
@deadprogram yes it is

aykevl, 2019-10-19:
I have a partial fix, will send a PR soon.

aykevl, 2019-10-31:
More focused repro:

```go
package main

import "reflect"

type Bar struct {
	A string
}

func main() {
}

func init() {
	println(reflect.TypeOf(Bar{}).Field(0).Name)
}
```

aykevl, 2019-10-31:
Retitling because this issue has nothing to do with WebAssembly but with the interp pass.

misterikkit, 2019-11-25:
I hit the same `panic: todo: store` when compiling a project, but my code isn't calling fmt.Sprint() anywhere I can see. Do you know how I can tweak my code to avoid hitting this error?

misterikkit, 2019-11-25:
> Do you know how I can tweak my code to avoid hitting this error?

Nevermind, figured out it was the interaction of globals & functions.

```
import "github.com/lucasb-eyer/go-colorful"

var (
    purple = func() color.RGBA { r, g, b := colorful.Hcl(320, 73.5, 50).RGB255(); return color.RGBA{r, g, b, 0} }()
    green  = func() color.RGBA { r, g, b := colorful.Hcl(124, 71.5, 78).RGB255(); return color.RGBA{r, g, b, 0} }()
)
```

deadprogram, 2019-11-26:
This should be resolved now in `dev`.

deadprogram, 2019-11-26:
This code has been released in v0.10.0 now closing.

PR #675 (2019-11-01): interp: add testing for scanning for side effects
I was trying to fix #627 but it turned out to be harder than expected. While doing that, I added some more testing to the interp package which is what this PR is about.
closed 2019-11-26 by deadprogram