tinygo-org/tinygo #632: Execute other executable

probonopd, 2019-10-17:
How can I execute another executable from TinyGo (Linux target) and act on its output/return code? (Not sure whether this is the correct place to ask, please advise.)

jadr2ddude, 2019-10-17:
At the moment we do not have support for that. Depending on your needs, you could probably use cgo for this, but you will not be able to access stdin and stdout at the same time yet. I am hoping to get some of the things required done, but it might be a bit before `os/exec` works.

probonopd, 2022-08-04:
Been almost three years, it seems a bit hard to understand that such basic functionality is missing.

Is there still a chance that `os/exec` will ever be supported?

deadprogram, 2024-11-27:
Closing since I think that https://github.com/tinygo-org/tinygo/pull/4377 and other similar work is accommodating this.