tinygo-org/tinygo #642: wasm_exec: publish to npm

darky, 2019-10-20:
In javascript community npm used for fetch javascript packages

aykevl, 2019-10-21:
Why?
The wasm_exec.js file is compiler specific and should always be copied from targets/wasm_exec.js. It may change with every TinyGo release. It doesn't make sense to put it in a dependency management system.

darky, 2019-10-21:
> The wasm_exec.js file is compiler specific

Why it reason not push it to npm?

> should always be copied from targets/wasm_exec.js

It's very annoying for js developer. 99,9% of packages I fetch from npm, but for `wasm_exec.js` need special way

> It may change with every TinyGo release

You can update `wasm_exec.js` for every TinyGo release.

Example of Golang `wasm_exec.js`: https://www.npmjs.com/package/@woodenfish/golang-wasm_exec

darky, 2019-10-23:
Ok, I think this task more related to JavaScript community instead of Golang community.
I think it can be resolved via thrird party person like this example https://www.npmjs.com/package/@woodenfish/golang-wasm_exec

closed 2019-10-23 by darky