tinygo-org/tinygo #644: wasm_exec: avoid use Go in global

darky, 2019-10-20:
Mutate global object, it's bad taste in javascript community.
Good case, to use commonjs or EcmaScript modules for this purposes.

aykevl, 2019-10-21:
Which global do you mean?

darky, 2019-10-21:
This place https://github.com/tinygo-org/tinygo/blob/8a771e3f89679e32aa1f7eeaf630fe1263b91803/targets/wasm_exec.js#L74

Would be better: `export class Go ...`

jccit, 2019-10-21:
Using ES modules here doesn't make much sense, since the wasm_exec.js is not expected to be ran through a bundler. I just tried to build it with webpack and wouldn't compile due to webpack tripping the `isNodeJS` variable.

darky, 2019-10-21:
Ok, I think refactor `wasm_exec.js` to proper format is unnecessary

closed 2019-10-21 by darky