tinygo-org/tinygo #645: WebAssembly: return non-number type on exported func

darky, 2019-10-21:
Some examples of calling exported functions from JavaScript

```go
//go:export test
func test() float64 {
  return 2.0 // It ok, return 2
}
```

```go
//go:export test
func test() bool {
  return true // No bad, return 1 and it truthy
}
```

```go
//go:export test
func test() string {
  return "some string" // Bad, return `undefined`
}
```

```go
//go:export test
func test() []float64 {
  var resp float64
  // ...Some population of resp
  return resp // Bad, return `undefined`
}
```

darky, 2019-10-23:
This issue inspired from `wasm-bindgen` cargo of Rust
Example: https://github.com/darky/rstar-wasm/blob/master/src/lib.rs

aykevl, 2019-10-23:
Yes, we'd like to support that eventually but nobody has taken up the work. If you'd like to work on it, that would be great!

deadprogram, 2019-10-25:
Hi @darky there are a couple of issues currently related to updating wasm_exec.js

https://github.com/tinygo-org/tinygo/issues/410
https://github.com/tinygo-org/tinygo/issues/631

Just wondering if updating to a newer version is called for before making the changes you propose... what do you think?

darky, 2019-10-25:
@deadprogram thanks. I consider this issues.

aykevl, 2019-10-25:
It is in fact impossible to directly return other types than integers and floats from WebAssembly functions because the spec doesn't allow it. However, it would be possible with bindgen wrappers.

Renaming this issue to reflect that.

darky, 2019-10-25:
I think it impossible to support `wasm-bindgen` from Rust.
Maybe proper need to create alternative like `wasm-bindgen`

aykevl, 2019-10-25:
As far as I know, wasm-bindgen was designed to be language independent.

tachyonicbytes, 2023-03-02:
Is this still being worked on?

I am trying to compile the following code with `-target wasi` and access it through `wasmtime` in `python`:

```go
package main

import "fmt"

//export Hello
func Hello(name string) string {
    return fmt.Sprintf("Hello, %s!", name)
	return name + 1
}

func main() {}
```

Is there any other way currently to produce an `example.wasm` file and to call the `Hello` function somehow? Just the `Hello` function, no `main`.

deadprogram, 2023-11-29:
This exists now https://github.com/bytecodealliance/wit-bindgen#guest-tinygo

Going to close this issue at long last... thanks everyone!

closed 2023-11-29 by deadprogram