tinygo-org/tinygo #646: Support Arduino Leonardo (ATmega32u4)

rgm3, 2019-10-22:
Please add support for the ATmega32u4 microcontroller as seen in the Arduino Leonardo.
https://www.arduino.cc/en/Main/Arduino_BoardLeonardo

I'd love to see some TinyGo USB HID APIs that worked on this target, but that's outside the scope for basic support.

aykevl, 2019-10-23:
I'd love to see support, but unfortunately I'm focusing on other targets at the moment (mainly ARM microcontrollers). The reason is that AVR (the instruction set architecture of the chip) has very weak support in LLVM, which is what TinyGo is based upon. I have seen many miscompilations, too many to be usable for bigger programs. ARM support is far more solid.

If you want to have it added, a pull request will be very welcome :)

deadprogram, 2019-10-25:
@rgm3 those are the same specs as the MakeyMakey board. If you do get around to a PR for the Leonardo, I will adapt it for the MakeyMakey.

jamesgoodhouse, 2023-02-14:
Sadness, I just picked up one of these and was hoping to use TinyGo. I'm guessing support is probably not something that's going to happen?

rgm3, 2023-02-14:
It looks like release 0.27.0 resolves this issue, although I haven't tried it myself yet. Thank you for your continued work on this fun project.

jamesgoodhouse, 2023-02-14:
I'll give it a whirl soon, although I'm 100% new to TinyGo, so I might not be the best to determine if the ATmega32u4 support is working as expected or not :-)

sago35, 2023-02-21:
Supported via TinyGo 0.27.
https://github.com/tinygo-org/tinygo/releases/tag/v0.27.0

This issue can be closed.
If you have any problems, please re-open it.

closed 2023-02-21 by sago35