tinygo-org/tinygo #648: Crash in interp

aykevl, 2019-10-22:
When running tinygo 0.9, a crash happens at this line:

https://github.com/tinygo-org/tinygo/blob/8a771e3f89679e32aa1f7eeaf630fe1263b91803/interp/values.go#L349

For this program:

https://github.com/hybridgroup/hacklab-2019/blob/master/sensor/arduino/step7/main.go

`tinygo flash -target arduino-nano33 main.go`

deadprogram, 2019-10-24:
I am unable to reproduce this:

```shell
$ tinygo flash -target arduino-nano33 ./sensor/arduino/step7/main.go 
error: failed to reset port /tmp/tinygo386440535/main.bin: opening port: no such file or directory
$ tinygo version
tinygo version 0.9.0 linux/amd64 (using go version go1.13)
```

deadprogram, 2019-11-26:
Unable to reproduce this with the latest `dev` branch. Closing, please reopen if needed.

closed 2019-11-26 by deadprogram

aykevl, 2019-11-26:
Now I see this issue again, I realize that it is a duplicate of #742.