tinygo-org/tinygo #649: incorrect clock rates on atsamd51

Infinoid, 2019-10-23:
The CPU clock rate on atsamd51 is incorrect, and unpredictable.

I set up the SysTick clock to generate ticks every 10ms, and made an interrupt handler that flips a GPIO pin every time it fires.  Then I probed that pin with a scope.

First time, I saw a square wave with 14ms between transitions.
I hit the reset button, then I saw a square wave with 6ms between transitions.
I reset it again, then I saw a square wave with 12ms between transitions.
I reset it again, then I saw a square wave with 23ms between transitions.
I reset it again, the program didn't start up at all.
I reset it again, then I saw a square wave with 8ms between transitions.

The CPU is not running at 120MHz/  PLL0 is settling at a different rate each time the chip is reset. 
 This makes it difficult to work with timing-sensitive peripherals, like neopixels.

The initClocks() function for atsamd51 has a bug.  I will submit a pull request with the fix.

Infinoid, 2019-10-23:
With pull request #650, the clock rate is consistent and correct.  My SysTick handler fires once every 10ms, as expected.

aykevl, 2019-10-23:
I tried it, but in my case it doesn't actually work. Before this PR, it blinks but too slowly (for me, always at the same speed though). After this PR, it doesn't do anything anymore. Also, I have to reset the chip manually to enter the bootloader which indicates that at least the USB system doesn't work either (perhaps the whole chip hangs, or USB is running at the wrong speed).

This is my code, with loads of nops left out for brevity:

```go
package main

import (
	"machine"
	"device/arm"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		wait()

		led.High()
		wait()
	}
}

func wait() {
	for i := 0; i < 1000*1000; i++ {
		// 60x nop
		arm.Asm("nop")
		arm.Asm("nop")
		arm.Asm("nop")
		// [...snip...]
	}
}
```

Infinoid, 2019-10-23:
It's interesting that yours is consistent.  Maybe I have other problems.  I'm using a Feather M4 Express board, with some board/chip support code that I've hacked up to get this far.

What rate is your chip running at?

Infinoid, 2019-10-23:
I was also using the v0.8.0 release to develop/test this fix, maybe something else changed since then.  I'll update and re-test and report back.

aykevl, 2019-10-23:
I don't have very precise measurements right now but it seems to be running at around 32MHz (30-35MHz, checked by timing the difference between the blinks).

aykevl, 2019-10-23:
Oh and I'm running this on top of the dev branch, using the ItsyBitsy M4.

> maybe something else changed since

I don't believe anything relevant has changed but I could be wrong.

Infinoid, 2019-10-23:
> Also, I have to reset the chip manually to enter the bootloader which indicates that at least the USB system doesn't work either (perhaps the whole chip hangs, or USB is running at the wrong speed).

I have also been resetting manually to enter the bootloader.  I just figured that was normal. :P

Infinoid, 2019-10-23:
Yep, current dev branch definitely behaves differently from v0.8.0.

I am compiling your program with `-target=itsybitsy-m4` and running it on my feather m4 express board.  (The "LED" pin is different between these two boards, but the GPIO is available and I plugged another LED into it.)

With v0.8.0, I see it running at around 64MHz, plus or minus 10MHz.  The rate changes when I reset.  My patch fixes it to run consistently at 120MHz.

With current dev, I see it running at 35MHz, consistently.  The rate does not change when I reset.  My patch causes the program to not run at all.

I found a typo that was introduced between v0.8.0 and v0.9.0.  Fixing that fixes the build for me.  I will update the pull request.

Infinoid, 2019-10-23:
Your test program was supposed to blink the LED once per second, yes?  60 million nops between bit flips, should be 2 bit flips per second at 120MHz.

With the dev branch plus the updated pull request, I am able to run your program and it works.  It blinks a little more slowly than once per second, I think because of bus timings, loop overhead, cache effects, etc.

I also tried my original SysTick timer program with current dev + pull request; I see it running once every 10ms, as expected.

Here's a scope screenshot, with the X axis set to 10ms per division:

[image]

deadprogram, 2019-10-24:
Perhaps the reason it may run more slowly is that the calculations here https://github.com/tinygo-org/tinygo/blob/dev/src/runtime/runtime_atsamd51.go#L237 and here are not correct https://github.com/tinygo-org/tinygo/blob/dev/src/runtime/runtime_atsamd51.go#L261 when the processor is run at the correct frequency?

Infinoid, 2019-10-24:
I don't know if the RTC is running at the correct rate or not.  I don't think either of our reproducer programs use the RTC at all.

deadprogram, 2019-10-24:
The runtime implementation of `time.Sleep()` uses the RTC.

Infinoid, 2019-10-24:
Makes sense, but we don't call time.Sleep().

deadprogram, 2019-10-24:
I misunderstood that "either of our reproducer programs" meant programs that you yourself were using for testing, as I had read it to mean the current TinyGo examples.

Infinoid, 2019-10-24:
Sorry about the confusion.  We've measured the slowdown in 2 ways:

- System Timer interrupts fire much less often than expected
- Loops of `nop` instructions run much slower than expected

Based on both of the above, I think the `atsamd51` clock rate is at around 35MHz.  The expected rate is 120MHz, so it is running at about 1/3 of the expected speed.

With the PR applied, I see System TImer interrupts fire at the expected rate, with high precision.  I see @aykevl's `nop` program blink the LED at about 90% of the expected rate.  I'm assuming the difference there is is due to bus effects, loop overhead and the like.

Infinoid, 2019-10-24:
And I'm open to suggestions for how to capture timing issues like this in a unit test. :)

deadprogram, 2019-10-24:
I think that https://github.com/tinygo-org/tinygo/pull/656 correct the RTC init. Combined with the code in #650 everything seems a lot tighter. Not sure how to test this as of yet... have to make that part of a future hardware CI setup, perhaps.

In any case, I think this issue can now be closed as the original issue has been addressed.

PR #650 (2019-10-24): runtime/atsamd51: fix clock init code
This fixes #649 .

The code is trying to set up the following chain:

GCLK7 = DFLL48M / 24 = 2MHz
DPLL0 = GCLK7 * 60 = 120MHz

But the setup of GCLK7 was wrong.  DIVSEL is a single-bit flag field, DIV is a 16 bit integer field.  The code should assign the value (24) to the integer field, not the flag.
closed 2019-10-24 by deadprogram