tinygo-org/tinygo #652: How can I build a go binary file with tinygo and the binary file can run on a embedded linux?

TonyAsian, 2019-10-24:
In my situation, there is a embedded linux OS on my ARM processor, so I cannot flash the binary code to the flash directly.

jadr2ddude, 2019-10-24:
So if you run the tinygo compiler on an ARM system, you can run:
```
tinygo build -o someProgram.elf my/package
```

Then `someProgram.elf` is a runnable linux executable.

TonyAsian, 2019-10-25:
@jadr2ddude Thank you! I'll try it.

aykevl, 2019-10-25:
Did you try setting the GOARCH variable? That should work.

```
GOARCH=arm tinygo build -o someProgram.elf my/package
```

You may need to install cross compilation tools (like `arm-none-eabi-gcc`).

deadprogram, 2020-07-05:
Closing this issue due to age and also that this question has been answered in other places. Please reopen if needed. Thanks everyone!

closed 2020-07-05 by deadprogram