tinygo-org/tinygo #655: board support: adafruit feather m4 express

Infinoid, 2019-10-24:
https://www.adafruit.com/product/3857
https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51

This board is very similar to the itsybitsy m4 board.  It is possible to run programs compiled for itsybitsy m4 unchanged on the feather m4 express.  If you connect an LED to the pin marked "12" on the board's edge, you can see the `blinky1` example program doing its divine work.

Some key differences:

- the on-board LED is on pin `PA23`, not `PA22`
- a NeoPixel (ws2812 RGB LED) is on pin `PB03`
- a 2MB flash storage device is attached to the `QSPI` pins
- uses a larger 64-pin variant of the `atsamd51`, which means it has `I²S` support

Infinoid, 2019-10-24:
I don't see any existing code for `QSPI` on this chip family.

The `I²S` registers on `atsamd51` look broadly similar to `atsamd21`.  The big difference is that the `CTRL` and `DATA` registers are not banked (numbered 0 and 1); one of the register sets is explicitly called `TX`, and the other is called `RX`.  Some of the register field layouts differ as well, but I don't think it will be hard to adapt.

aykevl, 2019-10-24:
If you want, you could implement this yourself. I would suggest copying and modifying the following files:

* targets/itsybitsy-m4.json -> targets/feather-m4-express.json
* src/machine/board_itsybitsy-m4.go -> src/machine/board_feather-m4-express.json

I don't have the hardware so can't do it myself.

aykevl, 2019-10-24:
Oh, it's a different chip? In that case, you can move the functions that differ into separate src/machine/machine_atsamd51*.go files.

Infinoid, 2019-10-24:
I'll take a stab at implementing this, but it will happen in a couple of stages.  The board and chip support files are easy, QSPI and I²S are more involved.

I'm looking to use I²S for a project, so I'm likely to tackle that before QSPI.

aykevl, 2019-10-25:
For initial board support, you don't need to support all features. Just basic ones like sleep and GPIO. But I guess they would all already work.

Once that's in, you can incrementally add support for all other features you'd like to see supported. In fact, I strongly prefer multiple small PRs over few large PRs because those are much easier to review.

deadprogram, 2019-10-28:
I would suggest add one PR with the board pin mapping, and then separate PRs for I2S etc.

This will make it a lot easier for others to possibly help out with the implementation and testing.

bgould, 2019-11-05:
Initial support for this board is here: https://github.com/tinygo-org/tinygo/pull/688

deadprogram, 2019-11-05:
Setting this issue to "Next release".

deadprogram, 2019-11-26:
This code has been released in v0.10.0 now closing.

closed 2019-11-26 by deadprogram