tinygo-org/tinygo #663: commands.go: doesn't use the right path for installation of clang-8

matloob, 2019-10-27:
I followed the instructions on the website to install tinygo. I'm using the ARduino Nano-33-IOT, which is an ARM device so I also installed LLVM.

```
brew tap tinygo-org/tools
brew install tinygo

brew install llvm@8
```

Then I ran the following command to flash the blinky example to my device:

```
tinygo flash -target=arduino-nano33 -port=/dev/cu.usbmodem14601 examples/blinky1
```

which produced the following error:

```
error: failed to build /usr/local/Cellar/tinygo/0.9.0/lib/compiler-rt/lib/builtins/absvdi2.c: fork/exec /usr/local/opt/llvm/bin/clang-8: no such file or directory
```

But when I installed llvm@8 using homebrew, the path it installed llvm at was `/usr/local/opt/llvm@8` not `/usr/local/opt/llvm`. I think (though I haven't tested this yet) that this could be fixed by updating the code [here](https://github.com/tinygo-org/tinygo/blob/8a771e3f89679e32aa1f7eeaf630fe1263b91803/commands.go#L23) to also add the /usr/local/opt/llvm@8 option as a location llvm could be installed to.

deadprogram, 2019-10-28:
Hi @matloob thanks for the research. I think you are correct about the solution needed here. Would you be able to try this out and submit a PR?

matloob, 2019-10-28:
I can prepare a PR, but I've had trouble building tinygo, so I can't test it:
```sh
# in the tinygo directory
$ go install
# github.com/tinygo-org/tinygo/cgo
cgo/libclang.go:18:10: fatal error: 'clang-c/Index.h' file not found
#include <clang-c/Index.h> // if this fails, install libclang-8-dev
         ^~~~~~~~~~~~~~~~~
1 error generated.
# tinygo.org/x/go-llvm
../../../../pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20190818154551-95bc4ffe1add/analysis.go:17:10: fatal error: 'llvm-c/Analysis.h' file not found
#include "llvm-c/Analysis.h" // If you are getting an error here read bindings/go/README.txt
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
```

aykevl, 2019-10-29:
Did you install the relevant LLVM packages? In particular, on Debian you'll need `clang-8`, `llvm-8-dev`, `lld-8`, and `libclang-8-dev`. For more information, see https://tinygo.org/getting-started/linux/

deadprogram, 2019-10-29:
This is a macOS issue with Homebrew path, I think.

aykevl, 2019-10-29:
I realized that right after posting the comment...
You will likely need to adjust the paths in cgo/libclang_config.go and make a similar change in [llvm_config_darwin.go](https://github.com/tinygo-org/go-llvm/blob/master/llvm_config_darwin.go) in the go-llvm package.

matloob, 2019-10-30:
Okay it seems to work. I'm wondering if we want to include both the  /usr/local/opt/llvm@8 and /usr/local/opt/llvm paths in the search paths or just the llvm@8 path. I'm guessing that some people do have their installation at /usr/local/opt/llvm so we want both? Though that means there will be warnings that one or the other path doesn't exist.

deadprogram, 2019-10-30:
OK I did a little more research, and I think there is a much simpler solution here that should not require any changes to TinyGo itself.

Homebrew has a `brew switch` command that is intended to use for this specific purpose:
https://docs.brew.sh/Manpage#switch-formula-version

You should be able to run TinyGo using the older version of LLVM by running this command like this:

```
brew switch llvm 8.0.1
```

The `brew switch` command changes the syminks used by Homebrew, so no further messing around with the PATH should be needed.

If you are not sure which LLVM version(s) you have already installed, you should be able to run:

```
brew info llvm
```

To see the specific versions.

aykevl, 2019-10-31:
I think adjusting the paths would be useful anyway, because then TinyGo is compatible with systems that have both LLVM versions installed. At the moment it requires another manual step to install TinyGo on macOS.

> Okay it seems to work. I'm wondering if we want to include both the /usr/local/opt/llvm@8 and /usr/local/opt/llvm paths in the search paths or just the llvm@8 path. I'm guessing that some people do [...]

I would propose only using the `llvm@8` paths. It seems like homebrew always creates that path to install LLVM to.

deadprogram, 2019-10-31:
>I think adjusting the paths would be useful anyway, because then TinyGo is compatible with systems that have both LLVM versions installed. 
The `brew switch` command is pretty designed for exactly this dual install scenario.
>I would propose only using the llvm@8 paths. It seems like homebrew always creates that path to install LLVM to.
Not sure we can always count on that naming, vs. using `brew switch`.

As you can tell I am in favor of using `brew switch`.

aykevl, 2019-10-31:
They are not mutually exclusive.

I would still suggest switching to `llvm@8` or having both. In the case of both paths, `brew switch` will also work.

deadprogram, 2019-10-31:
Sounds good to me!

deadprogram, 2019-11-04:
Code to address this has been merged into the `dev` branch as part of #676 now closing.

perttierkkila, 2019-11-05:
For me this didn't work, just says llvm not found (llvm@8 is installed)
> ```
> brew switch llvm 8.0.1
> ```

To fix llvm-symlinks, I had to run
```
brew link --force llvm@8
```

aykevl, 2019-11-05:
Can you paste the error output?

Also, to be sure, can you give the output of `tinygo version`?

perttierkkila, 2019-11-05:
Sure, I installed tinygo with
```
brew tap tinygo-org/tools
brew install tinygo
```

Tried to build led blinking example, got exactly same error than OP. Tried to fix with command
```
brew switch llvm 8.0.1
Error: llvm not found in the Cellar.
```

Found link-command somewhere when messing with versions. Forcing link fixed tinygo building, remarkable that still
```
ls -al /usr/local/opt/llvm/bin/clang-8
ls: /usr/local/opt/llvm/bin/clang-8: No such file or directory
```

```
brew info llvm
llvm: stable 9.0.0 (bottled), HEAD [keg-only]
Next-gen compiler infrastructure
https://llvm.org/
Not installed
```

```
brew info llvm@8
llvm@8: stable 8.0.1 (bottled) [keg-only]
Next-gen compiler infrastructure
https://llvm.org/
/usr/local/Cellar/llvm@8/8.0.1_1 (6,807 files, 3.3GB) *
```

```
tinygo version
tinygo version 0.9.0 darwin/amd64 (using go version go1.12.9)
```

deadprogram, 2019-11-06:
I see form the output that your install of LLVM is "key-only"
https://docs.brew.sh/FAQ#what-does-keg-only-mean

That would indicate why you needed to use the `brew link` command here.

deadprogram, 2019-11-06:
Reopening will close once the code in `dev` is released as part of next update.

aykevl, 2019-11-06:
>     tinygo version
>     tinygo version 0.9.0 darwin/amd64 (using go version go1.12.9)

The fix is not yet included in a TinyGo release, it will be included in the next 0.10.0 release (probably happening somewhere this month).

deadprogram, 2019-11-26:
This code has been released in v0.10.0 now closing.

closed 2019-11-26 by deadprogram