tinygo-org/tinygo #672: SPI should be a ReadWriteCloser

cosmotek, 2019-10-30:
Why make a custom SPI interface, why not use io.ReadWriteCloser? This interface would be far more useful IMO.

aykevl, 2019-10-31:
Because SPI does not have the same semantics.

SPI always does reading/writing at the same time. To read a byte, you also need to write a byte. And to write a byte, you'll always get a byte back (even if it's unnecessary). Most SPI-based protocols use either reading or writing at a time, but some need to do both at the same time. Therefore, just using `Read` or `Write` would not be sufficient. That said, it might be sensible to add `Read` and `Write` methods anyway for the majority of protocols that don't need read and write at the same time.

Compare this with UART: it implements the `io.ReadWriter` interface because reading and writing does not necessarily happen at the same time.

I don't think a `Close` method makes much sense. What should it do?
Many SPI peripherals support putting the peripheral to sleep. However, that would not fit with a `Close` method (as it is used on files) because you would have to resume the peripheral somehow in the future, which is not something that's possible with file descriptors.

deadprogram, 2019-11-06:
It would be nice to add a `ReadWriter` interface on top of the synchronous protocol, but it is not always clear what the semantics would be. One example where such as interface would be handy is SPI devices like the ESP32 where the communication protocol resembles the same AT protocol that is used with the UART interface to the same device. But if you examine the low level protocol details of that device, it has a lot more internal knowledge on the master side to know how to handle the synch. communication.

In any case, this is part of why SPI does not implement the `ReadWriter` interface as of yet. I think @aykevl has already mentioned why `Close()` probably does not make sense here.

cosmotek, 2020-06-25:
I kinda wonder if something like ReadWriteCloser could be implemented by buffering both the read and write until close is called (at which point the data would be communicated).

aykevl, 2020-06-30:
How could that work? Writing can indeed be buffered but reading cannot. Also, it would make the implementation very complex in a way that is very application specific. You are free to build such a wrapper around a `machine.SPI` type, however.

deadprogram, 2024-12-11:
Closing since no action required. Thanks everyone!

closed 2024-12-11 by deadprogram