tinygo-org/tinygo #678: function is already defined:internal/bytealg.CountString

aykevl, 2019-11-01:
To reproduce:

```
$ tinygo build -o test.elf -target=arduino-nano33 tinygo.org/x/drivers/examples/espat/mqttclient
../../../../../../github.com/tinygo-org/tinygo/src/runtime/string.go:223:6: function is already defined:internal/bytealg.CountString
$ go version
go version go1.12 linux/amd64
$ tinygo version
tinygo version 0.9.0 linux/amd64 (using go version go1.12)
```

deadprogram, 2019-11-01:
```shell
$ tinygo version
tinygo version 0.10.0-dev linux/amd64 (using go version go1.13)
$ go version
go version go1.13 linux/amd64
$ tinygo build -size short -o test.elf -target=arduino-nano33 ./examples/espat/mqttclient/main.go 
   code    data     bss |   flash     ram
  70448    8740    4600 |   79188   13340
```

bgould, 2019-11-03:
I hit this bug also with https://github.com/bgould/tinygo-model-m:

```
$ tinygo build -target=feather-m0 -o keyboard.uf2 main.go
../../../../../usr/local/tinygo/src/runtime/string.go:223:6: function is already defined:internal/bytealg.CountString
bcg@bcg-desktop:/tinygo/src/github.com/bgould/tinygo-model-m$ tinygo version
tinygo version 0.9.0 linux/amd64 (using go version go1.12.10)
```
Also I just tried upgrading to to Go 1.13 and it seems to have fixed the problem for me with no other changes to my TinyGo installation.

aykevl, 2019-11-03:
It happens only sometimes on older Go versions and is fixed by upgrading to Go 1.13. I'm going to close this as I don't think fixing it is worth the effort.

closed 2019-11-03 by aykevl