tinygo-org/tinygo #691: Judge type error when compiling go source code to wasm

elvin-du, 2019-11-05:
My cmds are as follows。
```
tinygo build -target wasm -o main.wasm main.go
# main.go
main.go:33:55: cannot use length (variable of type int) as uintptr value in struct literal
```

my go codes as follows:
```
//go:export Get
func Get() {
....
	length := len([]byte("dmx"))

	str := reflect.StringHeader{Data: uintptr(ptr), Len: length}
	.....
}
```

reflect.StringHeader definition as follows:
```
type StringHeader struct {
	Data uintptr
	Len  int
}
```

Len type is int。

jadr2ddude, 2019-11-05:
So the issues here is that the definition of `StringHeader` is different between Go and TinyGo at the moment. TinyGo defines `StringHeader` as:
```
type StringHeader struct {
	Data uintptr
	Len  uintptr
}
```

We may need to try to make some hacky workaround to get this to work. . .

aykevl, 2019-11-05:
See the documentation (emphasis mine):

>  StringHeader is the runtime representation of a string. **It cannot be used safely or portably and its representation may change in a later release.** Moreover, the Data field is not sufficient to guarantee the data it references will not be garbage collected, so programs must keep a separate, correctly typed pointer to the underlying data.

So technically the code that fails to compile is inherently unportable. However, the the only architecture where `int` and `uintptr` have a different size is on AVR. Perhaps we can change the type on non-AVR systems to be `int` for compatibility?

deadprogram, 2019-11-11:
I'm not sure if this is a bug exactly. What would be the correct category for this issue?

deadprogram, 2019-12-21:
What is the status of this issue/question?

deadprogram, 2020-05-13:
I am going to close this due to age and that it is somewhat ambiguous. Please reopen if needed. Thank you.

closed 2020-05-13 by deadprogram

sananguliyev, 2023-01-08:
I have the same issue: 

```
# github.com/cespare/xxhash
../../../go/pkg/mod/github.com/cespare/xxhash@v1.1.0/xxhash_unsafe.go:27:11: cannot use len(s) (value of type int) as uintptr value in assignment
../../../go/pkg/mod/github.com/cespare/xxhash@v1.1.0/xxhash_unsafe.go:28:11: cannot use len(s) (value of type int) as uintptr value in assignment
```

aykevl, 2023-01-11:
@sananguliyev please see https://tinygo.org/docs/guides/compatibility/#reflectsliceheader-and-reflectstringheader.