tinygo-org/tinygo #693: Odroid boards?

gdeverlant, 2019-11-05:
How can I use tinygo on Odroid boards?

aykevl, 2019-11-06:
Do you want to use TinyGo on the (Linux-based?) system that's on there, or do you want to compile a program to run baremetal on an Odroid board?

In the former case, just compile for arm/Linux. You can find instructions here: https://tinygo.org/getting-started/linux/

If you want to add support for running TinyGo programs baremetal, that will be significantly more work. Every revision of the Odroid appears to use a different SoC (often even from a different vendor), and TinyGo will have to be ported to each board specifically. You could take a look at this PR for adding Raspberry Pi 3 support for inspiration: https://github.com/tinygo-org/tinygo/pull/599

davecheney, 2026-10-26:
Thanks for the question. As considerable time has passed, I’m closing this issue. Please open a new issue if you have any questions about using TinyGo.

closed 2026-10-26 by davecheney