tinygo-org/tinygo #699: Wasm only compile error: function is already defined:internal/bytealg.CountString

justinclift, 2019-11-06:
With current TinyGo dev branch, using `strings.Split()` causes a compile failure when compiling to wasm:

```
$ tinygo build -target wasm -o foo main.go
../../../../../../../usr/local/go/src/internal/bytealg/count_generic.go:19:6: function is already defined:internal/bytealg.CountString
$ go version
go version go1.13.4 linux/amd64
```

The compile failure **only** happens when the target is `wasm`.  Without `-target wasm`, the compilation will succeed.

A minimal 100% reproducer:

```
package main

import (
	"strings"
)

func main() {
	sl := strings.Split("An example string", "\n")
	println(len(sl))
}
```

justinclift, 2019-11-10:
That reproducer also fails when run in the TinyGo playground, so it's not just me (wasn't sure). :smile:

aykevl, 2019-11-10:
I think this problem was introduced with the following PR: https://github.com/tinygo-org/tinygo/issues/424

justinclift, 2019-11-12:
Thanks @aykevl, #424 did fix this. :smile:

closed 2019-11-12 by justinclift