tinygo-org/tinygo #712: proposal: "machine/sync" package

jadr2ddude, 2019-11-11:
In order to use interrupts to implement wakeup from async operations, we need some high-level synchronization primitives to be available to the user. In the #606 experiment, a condition variable was added to a new package called `machine/sync`, as can be seen [here](https://github.com/jadr2ddude/tinygo/blob/d2b148340fd06f289943cb6acd205a47c5392222/src/machine/sync/cond.go).

Some remaining questions for this are:
1. What primitives are needed for common use cases?
2. What should this package be called (`machine/sync` was the first thing that came to mind, but it isn't necessarily that great)?

deadprogram, 2019-11-14:
What about just using aka actually implementing the standard `sync`? Do we need a separate package?

jadr2ddude, 2019-11-14:
This is different, so yes we need a separate package.

aykevl, 2019-11-19:
I would propose `runtime/sync`, or maybe even `runtime/interrupt`. The reason for putting it under the interrupt package is because I'd like to keep special cases for the compiler limited to the runtime package as much as possible.

jaddr2line, 2020-04-13:
In #1011, I went with `runtime/sync` as suggested

deadprogram, 2020-07-04:
Now that #1142 has been merged, I think we can mark this issue as "next release". Is this correct @jaddr2line ?

jaddr2line, 2020-07-04:
No, I think this issue is more closely related to interrupt-safe condition variables which it seems are going to need to be held back to v0.15