tinygo-org/tinygo #716: Wasm callbacks functions working with Go 1.12, but not working with Go 1.13

justinclift, 2019-11-12:
Found an interesting regression (?).  Callback creation code that works with TinyGo 0.9.0 + Go 1.12, doesn't work when used with Go 1.13.

For example, the canvas demo in @johanbrandhorst's [wasm-experiments](https://github.com/johanbrandhorst/wasm-experiments) repo:

* https://github.com/johanbrandhorst/wasm-experiments/blob/99f4e0a7f0ddf0215dd0eaae8f7f210767c5bba6/canvas/main.go#L36-L41
* https://github.com/johanbrandhorst/wasm-experiments/blob/99f4e0a7f0ddf0215dd0eaae8f7f210767c5bba6/canvas/main.go#L69

The error given on the browser console is:

```
panic: syscall/js: Value.Call: property _makeFuncWrapper is not a function, got undefined
```

Not sure what to err... make of this, as `_makeFuncWrapper()`  is a function included in the TinyGo wasm_exec.js, and is a literal copy of the [mainline Go](https://github.com/golang/go/blob/54e6ba6724dfde355070238f9abc16362cac2e3d/misc/wasm/wasm_exec.js#L550-L557) one.

Anyone have ideas?

OneOfOne, 2019-11-23:
Minimal repro:

```
package main

import (
	"syscall/js"
)

func main() {
	js.Global().Get("document").Call("onLoad", js.FuncOf(fn))
	select {}
}

func fn(this js.Value, args []js.Value) interface{} {
	return ""
}
```

justinclift, 2019-11-24:
Good thinking @OneOfOne.:smile:

ellemlabs, 2019-12-23:
Confirming. I'm using TinyGo (tag v0.11) with Go (v1.13.5) I get the same error. It doesn't seem to be exclusive to callbacks, but the use of any sort of javascript interaction in general:

```
package main

import "syscall/js"

func test(this js.Value, inputs []js.Value) interface{} {
	println("test")
	return nil
}

func main() {
	js.Global().Set("testfunc", js.FuncOf(test)) 
}
```

aykevl, 2019-12-30:
`js.FuncOf` essentially creates a callback, so yes it is still caused by callbacks.

OneOfOne, 2020-02-06:
1.14 is almost out, the issue is still there.

jfreymuth, 2020-02-14:
Line 335 in tinygo's `wasm_exec.js` doesn't exist in the mainline version. Removing that line makes the callback example work for me.

https://github.com/tinygo-org/tinygo/blob/1d913a62bc0ceef841953058c2c6fb1dd2a40f6a/targets/wasm_exec.js#L328-L337

justinclift, 2020-02-14:
That sounds hopeful. :smile:

tj, 2020-02-21:
I had luck with @jfreymuth's solution as well, although, I'm not sure what that breaks (if anything) haha

jfreymuth, 2020-02-23:
I'm rather certain this is the correct solution, the array in `wasm_exec.js` should match the definitions in the `syscall/js` package:
https://github.com/golang/go/blob/5bd145413a84be1afa74a82767384d9e224f7069/src/syscall/js/js.go#L101-L107
The change was made in commit https://github.com/golang/go/commit/c468ad04177c422534ad1ed4547295935f84743d (part of go 1.13).
Since `syscall/js` is not part of tinygo, changing our `wasm_exec.js` would break wasm for go 1.12 users. A possible solution would be for tinygo to ship it's own `syscall/js` package.

johanbrandhorst, 2020-02-23:
I think we have to try and stay compatible with the latest release of Go, so please submit a PR with the proposed changes and I'll review and merge it. @aykevl might have some further comments regarding go compatibility goals.

aykevl, 2020-02-23:
I've merged #911 to the dev branch, which means it will be part of the next release. Therefore I'll close the issue.

Thanks again @jfreymuth for discovering the issue and providing a fix.

closed 2020-02-23 by aykevl

justinclift, 2020-02-24:
> I think we have to try and stay compatible with the latest release of Go.

It's kind of a hard call when it's one of those "either/or" situations.  The bug this fixes is probably the only remaining thing (I can think of) which would be stopping people from using Go 1.13 with TinyGo anyway.  So should be safe in this instance. :smile: