tinygo-org/tinygo #725: Deb package does not set path to tinygo command

kgolding, 2019-11-15:
After a fresh install on Kubuntu 19.04, the tinygo command is not in the PATH and thus can't be easily run.

The proposal is to add a symlink like so `ln -s /usr/local/tinygo/bin/tinygo .`

deadprogram, 2020-03-16:
Since tinygo does not run well as a symlink, we need to handle this by either adding to PATH or making sure we document it.

@aykevl what do you think?

aykevl, 2020-03-16:
IIRC TinyGo can work just fine with a symlink. A symlink should be created by the .deb package, but I haven't looked into how to do that with `fpm`.

deadprogram, 2020-03-16:
At least one person ran into this problem recently at SCaLE who had installed the `deb package.

aykevl, 2020-05-13:
I have a fix in #1107.

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

PR #1107 (2020-05-13): ci: build .deb files along with .tar.gz files for Debian
This should make installing TinyGo easier on Debian based systems by automatically creating a symlink in `/usr/local/bin`. For example, see #1103 and #725. It also makes the release process a little bit more robust and reproducible as the Debian package is built in CI instead of on my laptop.

fixes #725
closed 2020-08-07 by deadprogram