tinygo-org/tinygo #726: crosspassage constants wont compile

ellemlabs, 2019-11-15:
`tinygo version 0.9.0 linux/amd64 (using go version go1.12.10)`

Well, this was a real head-scratcher to debug but after some investigation, I've managed to isolate a bug with the TinyGo compiler.

### Repository to use to Reproduce

To start, [I've made a repository for others to reproduce this error.](https://github.com/ellemlabs/tinygo-crosspackage-bug) See the comments in the main-test.go as well as the makefile for a quick insight.

### Description of Bug

When you assign a structure's field (defined in `package2`) to a constant (defined in `package1`), you get the following error even though both the structure's field and the constant are equal in type (defined in `package1`):

`cannot use package1.MyConst (constant 16777216 of type package1.MyInt) as package1.MyInt value in struct literal`

Note the bug can NOT be reproduced using the normal Go compiler. This leads me to conclude this is a TinyGo issue.

ellemlabs, 2019-11-15:
I've found a workaround. And that is to make an assignment function that removes the compiler's ability to type-check by using empty interfaces:

```
func WorkAround(mystruct *MyStruct, myfield interface{}) {
	myfieldFromInterface,_ := myfield.(package1.MyInt)
	mystruct.MyField        = myfieldFromInterface
}
```

aykevl, 2019-11-15:
That's a very strange bug. It appears to be a bug not in TinyGo directly but in the Go type checker. However, `go vet` doesn't complain which does make use of that same typechecker.

aykevl, 2019-11-15:
Looks like a bug in the handling of relative imports. Replacing all imports with normal Go-style imports works around the problem.

I suspect TinyGo imports one of the packages twice, thus leading to two "different" packages (and thus different types) for what is actually the same package.

ellemlabs, 2019-11-15:
@aykevl This what I was thinking. This error reminds me of the days when circular dependencies were an issue.

Perhaps the fact that both main-test.go and package2.go include package1.go results in package1 being imported twice.

aykevl, 2019-11-15:
Yes, that's my thinking.

In general I'd recommend using full import paths, like is common basically everywhere in the Go ecosystem. That should be a workaround for the bug. Not saying there isn't a bug in TinyGo, but this is an uncommon configuration and I'm not exactly sure what the right fix should be.

Unless this is a problem in a concrete case, I'm going to shelve this until TinyGo switches to [`"go/packages"`](https://godoc.org/golang.org/x/tools/go/packages) which will likely solve this issue too.

ellemlabs, 2019-12-18:
FYI: this is still a bug in `tinygo version 0.11.0-dev linux/amd64 (using go version go1.12.10)` (built from dev branch commit num 34ee388)

ellemlabs, 2019-12-21:
@aykevl If I wanted to attempt to fix this bug on my own, where would you recommend I start (source code wise)

aykevl, 2019-12-21:
It's in the [loader package](https://github.com/tinygo-org/tinygo/tree/dev/loader). It looks like the import path is used as the package path, which is usually the right thing to do, but for relative import paths I think the path should be normalized to be relative to the current working directory instead of to the importing package.

(But again, this loader package will eventually be replaced by x/go/packages or similar).

deadprogram, 2020-04-21:
Is this still an issue with the latest release, @aykevl ?

aykevl, 2020-04-26:
Nothing has changed in that regard. Eventually we'll want to use Go modules, which will probably fix this issue too.

deadprogram, 2020-06-11:
Now that we are using Go modules, is this issue still valid?

aykevl, 2021-01-13:
This is most likely fixed now. @ellemlabs please let us know if you're still having this issue.

closed 2021-01-13 by aykevl