tinygo-org/tinygo #727: isa<X>(Val) && "cast<Ty>() argument of incompatible type!"

ellemlabs, 2019-11-15:
`tinygo version 0.9.0 linux/amd64 (using go version go1.12.10)`

I'm compiling a larger project, about 10k lines of code into a WASM module. I have not compiled with TinyGo in some time (I switched to normal Go to do some testing). But after implementing many changes TinyGo's build is now failing with a hex dump.

It seems to be failing on the TinyGo -> LLVM level.

I will attempt to track down the problem by pealing away at my changes and isolating the error.
[errorreport.txt](https://github.com/tinygo-org/tinygo/files/3853519/errorreport.txt)

jadr2ddude, 2019-11-15:
Relevant line: https://github.com/tinygo-org/tinygo/blob/8a771e3f89679e32aa1f7eeaf630fe1263b91803/interp/values.go#L86

ellemlabs, 2019-11-15:
So far I've tracked it down to an included library, more specifically: github.com/OneOfOne/wjsu

Interestingly enough, the library does not import anything outside of "syscall/js" nor does it declare any constants... I'll continue to dig through.

aykevl, 2019-11-16:
This error is most likely caused by a global variable in one of the many packages. The "constant" it refers to is a different kind of constant as Go constants.

What I find interesting is that this build seems to be using the LLVM build with assertions enabled. It's nice to have for bugs like this, but it wasn't intended to be enabled. So I wonder how that got into a release build. @jadr2ddude any idea?

ellemlabs, 2019-11-16:
I've tracked the bug down [this exact line](https://github.com/OneOfOne/wjsu/blob/8a5720c9bca366918122d90b7c1c81ac9a405a5b/dom.go#L13): `var Document = HTMLDocument{HTMLElement{document}}` which is found in the global scope. If this line is commented out, or replaced with a getter, TinyGo builds it fine.

ellemlabs, 2019-11-16:
This code reproduces the error:

```
package main

import (
	"syscall/js"
)

var (
	document = js.Global().Get("document")
	head     = document.Get("head")

	Document = HTMLDocument{HTMLElement{document}}
)

type HTMLDocument struct {
	HTMLElement
}

type HTMLElement struct {
	v js.Value
}

func main() {

}
```

aykevl, 2019-11-17:
Even shorter reproducer:

```go
package main

func foo() byte

var array = [1]byte{foo()}

func main() {
}

```

aykevl, 2019-11-17:
This bug should be fixed with #735.

aykevl, 2019-11-24:
@ellemlabs the related PR has been fixed, can you check whether you still encounter this bug?

deadprogram, 2019-11-26:
This code has been released in v0.10.0 now closing.

closed 2019-11-26 by deadprogram

ellemlabs, 2019-12-20:
Confirm this works now. Good work.