tinygo-org/tinygo #728: LinkError: WebAssembly.instantiate(): Import #12 module="env" function="memcpy" error: function import requires a callable

ellemlabs, 2019-11-16:
After compiling a wasm file, if I attempt to invoke it using the typical means described in https://tinygo.org/webassembly/webassembly/, I get the error:

```
Uncaught (in promise) LinkError: WebAssembly.instantiate(): Import #12 module="env" function="memcpy" error: function import requires a callable
```

I'm not sure where to start with this. I believe it has something to do with the inner-workings of wasm_exec.js. I'm running the copy included with 0.9, but I've noticed it hasn't changed sense 0.8.

Regardless, is there anyway to debug this?

ellemlabs, 2019-11-16:
Using a `wasm2wat` and looking through the imports, it appears it's referencing to $memcpy, but that does not seem to be declared beforehand. I don't know too much about wasm or wat but this is certainly a start.

```
(module
  (type (;0;) (func (param i32)))
  (type (;1;) (func (result i32)))
  (type (;2;) (func (param i32 i32 i32) (result i32)))
  (type (;3;) (func (result f64)))
  (type (;4;) (func (param i32 i32 i32 i32 i32 i32 i32 i32 i32)))
  (type (;5;) (func (param i32 i32 i32 i32 i32 i32)))
  (type (;6;) (func (param i32 i32 i32 i32)))
  (type (;7;) (func (param i32 i32 i32 i32 i32)))
  (type (;8;) (func (param i32 i32 i32 i32 i32 i32 i32)))
  (type (;9;) (func))
  (type (;10;) (func (param i32 i32)))
  (type (;11;) (func (param i32) (result i32)))
  (type (;12;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;13;) (func (param i32 i32) (result i32)))
  (type (;14;) (func (param i32 i32 i32)))
  (type (;15;) (func (param i64 i32 i32) (result i64)))
  (type (;16;) (func (param i64) (result i32)))
  (type (;17;) (func (param i32 i64)))
  (type (;18;) (func (param i32 i32 i32) (result f64)))
  (type (;19;) (func (param i32 i32 i32) (result i64)))
  (type (;20;) (func (param i32 i64 i64 i64 i64 i64) (result i32)))
  (type (;21;) (func (param i32 i64 i64 i32 i64) (result i32)))
  (type (;22;) (func (param i32 i32 i32 i32 i32 i64 i32 i32)))
  (type (;23;) (func (param i32 i32 i32 i32 i32 i32 i32 i32 i32 i32 i32 i32)))
  (type (;24;) (func (param i32 i32 i32 i32 i32 i32 i32 i32 i32 i32 i32)))
  (type (;25;) (func (param i32 i32 i32 i32 i32 i32 i32 i32 i32 i32)))
  (type (;26;) (func (param i32 i32 i32 i32 i64 i32 i32)))
  (type (;27;) (func (param i32 i32 f64 i32 i32 i32)))
  (type (;28;) (func (param i32 i64 i32 i32)))
  (type (;29;) (func (param i32 i32 i32 i32 i32 i32) (result i32)))
  (type (;30;) (func (param i64 i32 i32 i32 i32) (result i64)))
  (type (;31;) (func (param i64) (result i64)))
  (type (;32;) (func (param i64 i32 i32 i32 i32)))
  (type (;33;) (func (param i32 i32) (result i64)))
  (type (;34;) (func (param f64) (result i64)))
  (type (;35;) (func (param i32 i64 i32 i32 i32 i32 i32)))
  (type (;36;) (func (param i32 i64 i32)))
  (type (;37;) (func (param i32 i64 i32 i32 i32)))
  (type (;38;) (func (param i32 f64 f64 i32 i32)))
  (type (;39;) (func (param i32 f64 i32 i32)))
  (type (;40;) (func (param i32 f64 i32 i32 i32)))
  (import "env" "io_get_stdout" (func $io_get_stdout (type 1)))
  (import "env" "resource_write" (func $resource_write (type 2)))
  (import "env" "runtime.ticks" (func $runtime.ticks (type 3)))
  (import "env" "syscall/js.valueCall" (func $syscall/js.valueCall (type 4)))
  (import "env" "syscall/js.valueGet" (func $syscall/js.valueGet (type 5)))
  (import "env" "syscall/js.valueSet" (func $syscall/js.valueSet (type 5)))
  (import "env" "syscall/js.valuePrepareString" (func $syscall/js.valuePrepareString (type 6)))
  (import "env" "syscall/js.valueLoadString" (func $syscall/js.valueLoadString (type 5)))
  (import "env" "syscall/js.stringVal" (func $syscall/js.stringVal (type 7)))
  (import "env" "syscall/js.valueNew" (func $syscall/js.valueNew (type 8)))
  (import "env" "syscall/js.valueLength" (func $syscall/js.valueLength (type 2)))
  (import "env" "syscall/js.valueIndex" (func $syscall/js.valueIndex (type 7)))
  ;; Import #12 module="env" function="memcpy" error: function import requires a callable
  (import "env" "memcpy" (func $memcpy (type 2))) 
```

ellemlabs, 2019-11-16:
Honestly, it's a miracle I've made it this far.

I've isolated the line of code that's causing this error and I have made the following code that can be used to reproduce the error:
```
package main

import (
	"crypto/sha256"
	"fmt"
)

func main() {
	h := sha256.New()
	h.Write([]byte("abc"))

       // This line prevents webassembly from linking properly and will cause the error
       // LinkError: WebAssembly.instantiate(): Import #... module="env" function="memcpy" error: function import requires a callable
	b := h.Sum(nil) 

	fmt.Printf("%v", b) // this line just makes sure the compiler doesn't cut any code.
}
```

ellemlabs, 2019-11-16:
I've tracked it down to the [exact line](https://github.com/golang/go/blob/6ba0be16391697bdaab7d8f836b66921977d5474/src/crypto/sha256/sha256block.go#L124) in the Go sha256 library:

```
// ...
p = p[chunk:]
// ...
```

The above instruction seems to inject a non-existing "$memcpy" call into the wasm binary:

```
   ;; ...
    local.get 3
    local.get 2
    i32.const 120
    call $memcpy ;; <- not a real function (should it be $runtime.memcpy?)
    local.set 2
    ;; ...
```

ellemlabs, 2019-11-16:
Welp. I've done it now.

After implementing the above sha256block.go file as referenced above into a local project, TinyGo refused to compile with an error (below). I started to cut away code until I got to the smallest possible instruction set to make the compiler break. And to my surprise, after cutting hundreds of lines I ended up with this:

```
package main
func main() {
	p := []byte{}
	for len(p) >= 1 {
		p = p[1:]
	}
}
```

Failing to compile because of this:

```
#$ /usr/local/tinygo/bin/tinygo build -target=wasm -o build/testt.wasm src/main-test.go
panic: stack chain start not found!

goroutine 1 [running]:
github.com/tinygo-org/tinygo/compiler.(*Compiler).makeGCStackSlots(0xc00000c3c0, 0x66f9900)
	/home/circleci/project/compiler/gc.go:198 +0x1951
github.com/tinygo-org/tinygo/compiler.(*Compiler).Optimize(0xc00000c3c0, 0x2, 0x2, 0x5, 0x0, 0x0)
	/home/circleci/project/compiler/optimizer.go:133 +0x7b6
main.Compile(0x7ffef8cb326f, 0x10, 0x7ffef8cb3255, 0x19, 0xc00000c1e0, 0xc000457e98, 0xc000457c68, 0x0, 0x0)
	/home/circleci/project/main.go:188 +0x370b
main.Build(0x7ffef8cb326f, 0x10, 0x7ffef8cb3255, 0x19, 0x7ffef8cb324d, 0x4, 0xc0000e5e98, 0xc00005c690, 0x201f1e1d1c1b1aff)
	/home/circleci/project/main.go:343 +0xe2
main.main()
	/home/circleci/project/main.go:836 +0x128d
```

I predict once this is resolved then the sha256 library will also start to work as well.

jadr2ddude, 2019-11-16:
I see what is happening here. Will try to fix.

ellemlabs, 2019-11-17:
Thanks @jadr2ddude 

Until then, a workaround is to edit the sha256block.go file's `blockGeneric` function to be this:

```
func blockGeneric(dig *digest, p []byte) {
	var w [64]uint32
	h0, h1, h2, h3, h4, h5, h6, h7 := dig.h[0], dig.h[1], dig.h[2], dig.h[3], dig.h[4], dig.h[5], dig.h[6], dig.h[7]
	for WorkAround1 := 0; WorkAround1 < len(p); WorkAround1 += chunk {
		WorkAround2 := p[kevin:]
	//for len(p) >= chunk {
		// Can interlace the computation of w with the
		// rounds below if needed for speed.
		for i := 0; i < 16; i++ {
			j := i * 4
			w[i] = uint32(WorkAround2[j])<<24 | uint32(WorkAround2[j+1])<<16 | uint32(WorkAround2[j+2])<<8 | uint32(WorkAround2[j+3])
		}
		for i := 16; i < 64; i++ {
			v1 := w[i-2]
			t1 := (v1>>17 | v1<<(32-17)) ^ (v1>>19 | v1<<(32-19)) ^ (v1 >> 10)
			v2 := w[i-15]
			t2 := (v2>>7 | v2<<(32-7)) ^ (v2>>18 | v2<<(32-18)) ^ (v2 >> 3)
			w[i] = t1 + w[i-7] + t2 + w[i-16]
		}

		a, b, c, d, e, f, g, h := h0, h1, h2, h3, h4, h5, h6, h7

		for i := 0; i < 64; i++ {
			t1 := h + ((e>>6 | e<<(32-6)) ^ (e>>11 | e<<(32-11)) ^ (e>>25 | e<<(32-25))) + ((e & f) ^ (^e & g)) + _K[i] + w[i]

			t2 := ((a>>2 | a<<(32-2)) ^ (a>>13 | a<<(32-13)) ^ (a>>22 | a<<(32-22))) + ((a & b) ^ (a & c) ^ (b & c))

			h = g
			g = f
			f = e
			e = d + t1
			d = c
			c = b
			b = a
			a = t1 + t2
		}

		h0 += a
		h1 += b
		h2 += c
		h3 += d
		h4 += e
		h5 += f
		h6 += g
		h7 += h
		//p = p[chunk:]
	}

	dig.h[0], dig.h[1], dig.h[2], dig.h[3], dig.h[4], dig.h[5], dig.h[6], dig.h[7] = h0, h1, h2, h3, h4, h5, h6, h7

}
```

All this does is change the mechanic it uses to loop. No performance difference as far as I can tell. Just less "Go"-y and more "C"-y.

aykevl, 2019-11-17:
That second error is actually a very different error.

The first error should be relatively simple to fix, with a fix very similar to this one: https://github.com/tinygo-org/tinygo/pull/137
LLVM can insert such calls depending on the optimization level. I'm aware of three of them: `memset`, `memcpy` and `memmove`.

deadprogram, 2019-11-26:
This issue should be fixed in `dev` @ellemlabs if you would like to verify.

deadprogram, 2019-11-26:
This code has been released in v0.10.0 now closing.

closed 2019-11-26 by deadprogram