tinygo-org/tinygo #733: panic: trying to make exported function async

ellemlabs, 2019-11-17:
I'm back with more weekend-ruining excitement.

I put a channel receive inside a method that cannot be directly called from the javascript. When compiling I get the following error. I notice that if I start putting other channel receives inside of arbitrary functions than I can't reproduce the error.

I'm not sure if this is a bug or the compiler is trying to tell me to stop what I'm doing.

```
panic: trying to make exported function async: (StructName).FunctionName$invoke

goroutine 1 [running]:
github.com/tinygo-org/tinygo/compiler.(*Compiler).markAsyncFunctions(0xc0001421e0, 0xc000000180, 0xc000036000, 0xc0064ff0d0)
	/home/circleci/project/compiler/goroutine-lowering.go:852 +0x5ae6
github.com/tinygo-org/tinygo/compiler.(*Compiler).lowerCoroutines(0xc0001421e0, 0xa9bf6b, 0x7fc8e8090a90)
	/home/circleci/project/compiler/goroutine-lowering.go:180 +0x40
github.com/tinygo-org/tinygo/compiler.(*Compiler).LowerGoroutines(0xc0001421e0, 0x7fc8e8091580, 0x7fc8e808f360)
	/home/circleci/project/compiler/goroutine-lowering.go:131 +0x6d
github.com/tinygo-org/tinygo/compiler.(*Compiler).Optimize(0xc0001421e0, 0x2, 0x2, 0x5, 0x0, 0x0)
	/home/circleci/project/compiler/optimizer.go:83 +0x5b7
main.Compile(0x7ffdf414c21f, 0x10, 0x7ffdf414c205, 0x19, 0xc000142000, 0xc0064ffe98, 0xc0064ffc68, 0x0, 0x0)
	/home/circleci/project/main.go:188 +0x370b
main.Build(0x7ffdf414c21f, 0x10, 0x7ffdf414c205, 0x19, 0x7ffdf414c1fd, 0x4, 0xc0000e7e98, 0xc0000ee250, 0x8cd78e)
	/home/circleci/project/main.go:343 +0xe2
main.main()
	/home/circleci/project/main.go:836 +0x128d
```

jadr2ddude, 2019-11-17:
This appears to just be #514 .

ellemlabs, 2019-11-17:
Do you suppose https://github.com/tinygo-org/tinygo/pull/736/commits/0e036d4fa9c88ab63f2ecb1a7746ede23af4f065 will fix it?

jadr2ddude, 2019-11-17:
No, but the main commit in that PR should fix it.

jadr2ddude, 2019-11-17:
This should be fixed on `dev`.

ellemlabs, 2019-11-18:
After changing some code around, and building and using `tinygo version 0.10.0-dev linux/amd64 (using go version go1.12.10)` built from `98eee7` I'm now getting this error:

```
error: async function src/main-wasm.go.FunctionName$1 used as function pointer
```

jadr2ddude, 2019-11-18:
Can you please provide some basic information on how you are invoking that function?

ellemlabs, 2019-12-20:
Unfortunately, I cannot track this one down. I can tell you that the error came from using `wjsu.Console.log("test")` in a callback function. In that wjsu library, every time you call 'Console.Log(...)' it passes the arguments through a "safe args" test which ultimately ends up going through a [function that includes a ton of reflection and implicit typing](https://github.com/OneOfOne/wjsu/blob/0b1fc81e4a3f8748352468c9431aef8994ae210b/other.go#L24).

After further testing, I've found that in most instances it works fine. Sometimes it doesn't compile with the error specified above. Other times the code seems to cease execution without any error or notification (almost like the browser kills the execution because of an unknown error).

Sorry I cannot provide more details. I'll dig into this into the future.

aykevl, 2019-12-21:
@ellemlabs can you take a look at https://github.com/tinygo-org/tinygo/pull/794? It should improve error reporting in this case, possibly making it clearer where the error is coming from.

ellemlabs, 2019-12-21:
@aykevl 

After failing to build with new code was making, it was sending an error "async function ... used as function pointer". I pulled and merged both of those commits you referenced and it indeed tracked the error down to the original callback. Furthermore, I found that the error only happened when I used the "time" library. I hypothesize that wasm builds fail when the compiler stumbles upon asm code.

Anyways, here's the most minimal code I have that reproduces this type of error. Let me know if it does not reproduce on your side.

```
package main

import (
        "syscall/js"
        "time"
)

func main() {
        js.Global().Call("setTimeout", js.FuncOf(func(this js.Value, args []js.$
                println(time.Now().Day())
                return nil
        }), 2000)
}
```
And the build:
```
$ tinygo build -target=wasm -o main.wasm main.go 

../../../usr/local/go/src/syscall/js/func.go:90:13: async function main.go.main$1 used as function pointer
```

ellemlabs, 2019-12-22:
Update: even smaller code. It appears gocallbacks are invalid when in the context of a javascript callback.
```
package main

import (
	"syscall/js"
)

func do(fun func ()) {
	fun()
}

func main() {
	js.Global().Call("setTimeout", js.FuncOf(func(this js.Value, args []js.Value) interface{} {
		//println(time.Now().Day())
		do(func() {
			println("asdf")
		})
		return nil
	}), 2000)
}
```

jaddr2line, 2020-03-27:
This now crashes with a nil pointer panic in `syscall/js.handleEvent`.

deadprogram, 2020-07-05:
What is the status of this issue now, is it still valid?

aykevl, 2020-11-04:
I think I have found this bug while working on #1430, that PR contains a commit that should fix this issue.

deadprogram, 2021-11-06:
Seems this issue has been fixed for some time. Now closing, please reopen if needed. Thanks!

closed 2021-11-06 by deadprogram