tinygo-org/tinygo #734: compiler-rt: docker dev image no longer working as expected

deadprogram, 2019-11-17:
Ever since the refactoring of compiler-rt the `tinygo-dev` docker image appears to longer work as expected. This issue was surfaced via the CI build in the Timygo drivers repo:

https://app.circleci.com/jobs/github/tinygo-org/drivers/269

```
tinygo build -size short -o ./build/test.elf -target=itsybitsy-m0 ./examples/adxl345/main.go
clang: error: no such file or directory: '/tinygo/lib/compiler-rt/lib/builtins/absvdi2.c'
```

aykevl, 2019-11-17:
I see a few potential solutions to this:

1. Revert https://github.com/tinygo-org/tinygo/pull/719.
2. Do `make llvm-source` in the Dockerfile
3. Do a `git clone` of that repository in the Dockerfile

Maybe it's best to just revert it. It leads to a bit of duplication, but it also fixes `go get` of the tinygo repository (it would otherwise have the same problem).

deadprogram, 2019-11-17:
OK reverting sounds like the least problematic. Will you do this?

deadprogram, 2019-11-25:
Reminder that we need to revert https://github.com/tinygo-org/tinygo/pull/719

deadprogram, 2019-11-25:
This has now been addressed in `dev`. Closing.

closed 2019-11-25 by deadprogram