tinygo-org/tinygo #747: How to use serial communication on other ports in Arduino?

cyantarek, 2019-11-23:
Hi,

How can I use other ports like port 7 & 8 for serial communication on android?
Very much like the SoftwareSerial C library.

aykevl, 2019-11-24:
As you mentioned, that requires software emulation of UART. I'm not aware of a driver for that, it would be a nice addition to the [drivers](https://github.com/tinygo-org/drivers) repository.

cyantarek, 2019-11-25:
If TinyGo doesn't support these library functionalities like "SoftwareSerial", "Ethernet", "Wifi", "GSM" etc, then the usage scope of TinyGo is very limited on the Arduino platform, as far as I can see.

aykevl, 2019-11-25:
Yeah I know. Because the AVR backend in LLVM is very much experimental and has lots of bugs, I haven't done much work on Arduino Uno support. However, there are many other chips (like the Arduino Nano 33 IoT and many boards from Adafruit) that have great support. Not with all those libraries, but there are many already and you're welcome to help with that.

cyantarek, 2019-11-25:
Alright.

I have found an example for Serial in tinygo/src/examples/serial/serial.go
Here's the code:

```
package main

import "time"

func main() {
	for {
		println("hello world!")
		time.Sleep(time.Second)
	}
}
```

It looks like println() function is writing data to the USB Serial Monitor. But I am not understanding how it's writing to the 0, 1 RX and TX Pin on the Arduino as there's no explanation in the code. May be it's a starting point. Can you tell me this please?

BattleBas, 2020-06-06:
I thought maybe serial communication could be done using UART where you can define a custom TX and RX but I haven't had any luck. Was hoping to get dfminiplayer working quick snippet below to get an idea what I thought would work:
```
sound:= machine.UART0
sound.Configure(machine.UARTConfig{TX: machine.Pin(2), RX: machine.Pin(3)})
sending := []byte{0x7E, 0xFF, 6, 3, 1, 0, 1, 0xFE, 0xF6, 0xEF}
sound.Write(sending)
```

aykevl, 2020-06-06:
> It looks like println() function is writing data to the USB Serial Monitor. But I am not understanding how it's writing to the 0, 1 RX and TX Pin on the Arduino as there's no explanation in the code. May be it's a starting point. Can you tell me this please?

`println` uses the serial output defined in the runtime, which varies by chip but is usually the first (or only) UART available. For example, for the Arduino Uno it's defined here:

https://github.com/tinygo-org/tinygo/blob/c5a896771d865fc77fac3e50c98e3d0ec98bad54/src/runtime/runtime_avr.go#L62-L68

You can also use this UART directly, see `src/examples/echo` for example.

> I thought maybe serial communication could be done using UART where you can define a custom TX and RX but I haven't had any luck.

For which board/chip?
On the Arduino Uno, there is only one hardware UART on fixed pins (pin 0 and 1). I'm not aware of a software serial library for TinyGo (but it could be written, of course) which would support other pins. Or if you really need more than one UART, you could try a different board which has hardware support for multiple UARTs.

BattleBas, 2020-06-06:
Thank you very much for the response and pointing out that the arduino has fixed pins for UART that helped resolve my issue (I am using an arduino nano). I was trying to use other pins thinking that a software serial library magically already existed. A software serial library would definitely be a cool feature in the future! 

Sorry for hijacking this issue haha

deadprogram, 2024-12-07:
Closing due to age, please reopen if needed. Thanks!

closed 2024-12-07 by deadprogram