tinygo-org/tinygo #752: Add Runtime Error Interface

bgokden, 2019-11-25:
I would like to add Runtime Error interface defined here: https://golang.org/pkg/runtime/#Error
to solve the problem of compiling text/template/parse as defined here: https://tinygo.org/lang-support/stdlib/#text-template-parse

I am testing some wasm libraries and they include text-template-parse. Adding this error interface may solve the problem.

aykevl, 2019-11-26:
That would be a welcome improvement! You could add it to src/runtime/panic.go.

For the `runtime.Error` type, see: https://golang.org/ref/spec#Run_time_panics

Note that TinyGo doesn't currently use such a type and honestly there isn't really a good reason to until it supports the `recover()` builtin. So the main benefit to having this type would be to get existing code to compile.

yuriizinets, 2021-11-26:
I understand that it's not `tinygo` main domain, but I'm exploring possibility of using this compiler to build WASM modules for edge workers. And I'm really missing `html/template`. Maybe there is some ETA to bring this standard lib into `tinygo`?

dgryski, 2021-11-26:
`text/template` relies heavily on `reflect` which is one of tinygo's weak points. You might be better off looking for a code-generation template library if possible.

yuriizinets, 2021-11-26:
Ok, I understand. Many thanks for reply!

deadprogram, 2024-12-07:
This was addressed quite some time back so now closing. Thanks everyone!

PR #1099 (2020-06-23): extend stdlib to allow import of more packages
Closes #752

these changes allow a successful import of the following go 1.14 stdlib packages:

```
	_ "database/sql"
	_ "debug/elf"
	_ "debug/macho"
	_ "debug/pe"
	_ "debug/plan9obj"
	_ "html"
	_ "mime"
	_ "testing"
```
closed 2024-12-07 by deadprogram