tinygo-org/tinygo #753: Can we use tinygo in ESP32 boards?

maheshjena1, 2019-11-26:
Can we use tinygo in ESP32 boards?

aykevl, 2019-11-26:
Not yet, unfortunately. For that we'd need an up-to-date version of LLVM with the Xtensa backend. For details: https://tinygo.org/faq/what-about-esp8266-esp32/

Closing as there is not much we can do about this.

closed 2019-11-26 by aykevl

maheshjena1, 2019-11-27:
thanks for the info.

racerxdl, 2019-11-27:
@aykevl according to https://www.esp32.com/viewtopic.php?t=9226, the LLVM / Clang works now with XTensa.

```
Updated 2019/03/26: support for both ESP32 and ESP8266 targets (-mcpu=esp32, -mcpu=esp8266)
```

deadprogram, 2019-11-27:
Actually https://github.com/espressif/llvm-xtensa has stalled for quite a while, sadly... hopefully it will pick back up again.

aykevl, 2019-11-28:
To add to that: we really need an up-to-date LLVM version to get it to work, because TinyGo depends on a recent stable release of LLVM (currently LLVM 9). While it would theoretically be possible to take Espressifs fork and rebase it upon LLVM 9, that would be a massive amount of work and would also make upgrading to a more recent LLVM version significantly harder. The only solution to that is upstreaming the backend, but work on that has sadly stalled.

I would really want to add ESP32 support, but until it's upstream I don't see how we can reasonably support it.

racerxdl, 2019-11-29:
That does make sense @aykevl. Let's see if Espressify will keep the work on LLVM.

manikawnth, 2020-02-03:
FYI.
The project has been archived and looks like active work is going on to commit to upstream llvm.
Here's the new repo to track

https://github.com/espressif/llvm-project

deadprogram, 2020-02-03:
Thanks for that info @manikawnth sounds very encouraging.