tinygo-org/tinygo #768: cannot build in Ubuntu WSL: clang-9 missing

misterikkit, 2019-12-04:
tinygo v0.10.0 adds a dependency on clang-9 which is not available to Ubuntu bionic distributions.

In fact, clang-9 seems to only be available in [eoan](https://packages.ubuntu.com/search?keywords=clang-9). On top of that, Ubuntu eoan is not yet supported in WSL. (Listed as [still in development](https://wiki.ubuntu.com/WSL#Installing_Ubuntu))

tinygo installed like so,
```
$ wget https://github.com/tinygo-org/tinygo/releases/download/v0.10.0/tinygo_0.10.0_amd64.deb
$ sudo dpkg -i tinygo_0.10.0_amd64.deb
```

Sample output:
```
$ tinygo build -o /mnt/e/flash.uf2 -target=itsybitsy-m4
error: failed to build src/device/arm/cortexm.s: none of these commands were found in your $PATH: clang-9
```

Windows Version	10.0.18362 Build 18362
Ubuntu Version
Description:    Ubuntu 18.04.3 LTS
Release:        18.04
Codename:       bionic

misterikkit, 2019-12-04:
Found these instructions to manually install clang-9, which seems to have me building again.

https://solarianprogrammer.com/2017/12/13/linux-wsl-install-clang-libcpp-compile-cpp-17-programs/

jaddr2line, 2019-12-04:
see: https://apt.llvm.org/ which is a common way to use LLVM

aykevl, 2019-12-04:
See the following section of the getting started docs: https://tinygo.org/getting-started/linux/#arm-cortex-m

misterikkit, 2019-12-09:
I was caught off guard by switching from 0.9 to 0.10.
If this is a documented step with working instructions, then I think this issue should be closed. Thanks for the pointers!

aykevl, 2019-12-10:
Ah, that's important context. Do you think there is something we can improve in the documentation to make this more clear?

For the next release, we'll hopefully drop the `clang-9` dependency entirely in release builds (it will be bundled with TinyGo) so you will hopefully not have this problem again.

misterikkit, 2019-12-10:
I think I'm just used to golang 1.0 compatibility, so even though tinygo is
0.x, I didn't think to look for release notes.

deadprogram, 2019-12-23:
TinyGo v0.11 was just released which no longer requires the separate Clang-9 install. Now closing, please reopen if needed. Thanks!

closed 2019-12-23 by deadprogram