tinygo-org/tinygo #770: unexpected token ILLEGAL

xaionaro, 2019-12-04:
```
xaionaro@ubuntu:~/go/src/github.com/u-root/u-root/bb$ CGO_ENABLED=0 tinygo build -o bb main.go 
# os/user
../../../../../../.gimme/versions/go1.13.linux.amd64/src/os/user/cgo_lookup_unix.go:19:6: not implemented: build constraints in #cgo line
/usr/include/x86_64-linux-gnu/bits/confname.h:222:31: unexpected token ILLEGAL
/usr/include/x86_64-linux-gnu/bits/confname.h:224:31: unexpected token ILLEGAL
/usr/include/x86_64-linux-gnu/bits/confname.h:222:31: unexpected token ILLEGAL
/usr/include/x86_64-linux-gnu/bits/confname.h:224:31: unexpected token ILLEGAL
/usr/include/x86_64-linux-gnu/bits/confname.h:222:31: unexpected token ILLEGAL
/usr/include/x86_64-linux-gnu/bits/confname.h:224:31: unexpected token ILLEGAL
```

```
xaionaro@ubuntu:~/go/src/github.com/u-root/u-root/bb$ head -224 /usr/include/x86_64-linux-gnu/bits/confname.h | tail -4
    _SC_GETGR_R_SIZE_MAX,
#define	_SC_GETGR_R_SIZE_MAX		_SC_GETGR_R_SIZE_MAX
    _SC_GETPW_R_SIZE_MAX,
#define	_SC_GETPW_R_SIZE_MAX		_SC_GETPW_R_SIZE_MAX
```

deadprogram, 2019-12-21:
Hi @xaionaro could you please provide the code you were trying to compile? Thank you.

aykevl, 2019-12-21:
Sorry, I looked into this but forgot to get back to you.

At the moment, these typedefs (containing symbol names) are not yet supported in CGo. They should be added eventually, though.

aykevl, 2019-12-30:
Here is a partial fix: https://github.com/tinygo-org/tinygo/pull/822

dkegel-fastly, 2021-09-10:
I seem to have run into this with tinygo dev building my wasi app on mac:
```
# os/user
/usr/local/Cellar/go/1.17/libexec/src/os/user/cgo_lookup_unix.go:21:6: not implemented: build constraints in #cgo line
```

This was triggered by adding a dependency, but I can't quite identify which source file from which dependency is importing os/user.  (go mod graph doesn't seem to help?)

closed 2024-11-27 by deadprogram