tinygo-org/tinygo #771: How to disable CGO?

xaionaro, 2019-12-04:
Hello.

I need to set something like `CGO_ENABLED=0`. There's some code that has different implementations for cases if CGo is supported and if CGo is not supported. And I need to compile the second variant. It would be nice if you would help me to understand how can I do it without manually modifying all my dependencies (which could use CGo)?

deadprogram, 2019-12-21:
I'm not sure you can compile a TinyGo program without CGo support at this time. @aykevl can confirm or correct that.

aykevl, 2019-12-21:
That's correct, CGo support cannot currently be disabled.

Ideally I hope to make CGo "just work" (more reliably than the standard Go toolchain), but we're not there yet. Until that time, adding support for `CGO_ENABLED` seems sensible.

aykevl, 2019-12-30:
@xaionaro can you test this PR? https://github.com/tinygo-org/tinygo/pull/821

xaionaro, 2020-01-01:
@aykevl : As experiment I've tried to make similar changes within tinygo (as in the PR) and it didn't help. But I was hurrying so I suppose I just missed something. Yes, I will test it within a week. Thank you :)

xaionaro, 2020-01-13:
Sorry for the late answer. It works. Thank you.

PR #821 (2020-01-14): compiler: add support for CGO_ENABLED environment variable
This should fix #771. Not sure how to test it, but you can test locally by running tinygo with the ./testdata/cgo package with the following extra file (main2.go):

```go
// +build !cgo

package main

func main() {
    println("no cgo")
}
```

and run with `CGO_ENABLED` set to `1` or `0` to test the difference.
closed 2020-01-21 by xaionaro