tinygo-org/tinygo #772: can't build it with neither system llvm nor source

OneOfOne, 2019-12-05:
```
┏━ oneofone@voyager ❰✪/t/tinygo❱ ❰master|✚1❱
┗━❰2❱━● make tinygo
CGO_CPPFLAGS="-I/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-project/llvm/include -I/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-build/include -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-project/clang/include -I/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++11" CGO_LDFLAGS="/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-build/lib/libclang.a -std=c++11 -L/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-build/lib -Wl,--start-group -lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -Wl,--end-group -lstdc++ -Wl,--start-group -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML -Wl,--end-group -L/home/oneofone/code/go/src/github.com/tinygo-org/tinygo/llvm-build/lib  -lLLVMOption -lLLVMMCJIT-lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVCodeGen -lLLVMRISCVAsmParser -lLLVMRISCVDesc -lLLVMRISCVUtils -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMGlobalISel -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMDebugInfoDWARF -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMX86Desc -lLLVMObject -lLLVMMCParser -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMMCDisassembler -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBinaryFormat -lLLVMX86Utils -lLLVMX86Info -lLLVMSupport -lLLVMDemangle -lrt -ldl -lpthread -lm -lxml2 -lstdc++ " go build -o build/tinygo -tags byollvm .
# github.com/tinygo-org/tinygo/interp
interp/errors.go:84:13: inst.InstructionDebugLoc undefined (type llvm.Value has no field or method InstructionDebugLoc)
```

aykevl, 2019-12-05:
Can you share the following information?

```
go version
echo $GO111MODULE
```

It looks like there is a problem with your version of the tinygo.org/x/go-llvm package, which suggests that you're not using Go modules.

OneOfOne, 2019-12-05:
Yep fixed.

closed 2019-12-05 by OneOfOne