tinygo-org/tinygo #777: Expand board and device support for stm32f4discovery and peripherals

gwtnz, 2019-12-08:
The stm32f4discovery board has a number of devices on it (MEMS microphone, I2S sound device, accelerometer, etc) that currently don't have support in tinygo, nor is there support for:
* SPI
* I2C
* PWM
* I2S
on the chip.

The machine_stm32f407.go UART peripheral code style differs in from that in the machine_stm32f103xx.go file, and while perhaps being closer to a C-developer's natural low-level usage of the chip, programmatically seems like a more complex implementation making expansion for the other peripherals less straightfoward.

I propose combining the two styles and moving them up to the machine_stm32.go file and then handling chip- and board-specific nuances in the chip family files.

deadprogram, 2019-12-08:
It would be great to share code and make the implementations more consistent. I am in favor of this proposal.

gwtnz, 2019-12-08:
I'm happy to take a shot at a PR

deadprogram, 2020-04-14:
This has now been released, so closing. Thanks!

closed 2020-04-14 by deadprogram