tinygo-org/tinygo #783: proposal: structure of gameboyadvance functionality

kylelemons, 2019-12-15:
We've discussed this a bit in the Gophers slack, but I haven't had much time to actually sit down and do it, so I think it's worthwhile to write down what I think the outcome of the discussions should be both so that I don't forget but also to have it on record for comment and other contributions.

## tinygo-org/tinygo

The functionality that would go in the core of tinygo would be:

- Changes to the compiler, linker scripts, or `tinygo build`
  - ROM preparation (e.g. `gbafix` functionality)
  - Flash / memory layout (e.g. specifying values to be stored in .rodata on flash cart)
- Pieces that may be required to implement runtime features
  - Timers
  - Interrupts (pending #515)
  - DMA
- Any provisions for System or ARM mode (rather than User and Thumb)
- Basics for "hello world"
  - Simple "Mode 3" [drivers.Displayer](https://godoc.org/tinygo.org/x/drivers#Displayer)

## tinygo-org/drivers/gba

The functionality that would go into a `gba` subdirectory within the drivers repo would be:

- GBA peripheral drivers
  - Video controller
    - backgrounds
    - sprites
    - palettes
    - windowing
  - Sound controller
  - Gamepad

An important aspect of this per discussion is that these drivers would contain most of the memory-mapped IO pieces without exposing them as part of the public API (e.g. via the `machine` package) as would be more common in other boards.  There are two primary reasons for this: (1) the GBA has a number of interleaved structs or otherwise complicated memory layouts that can be difficult to use correctly and which may make sense to evolve over time, and (2) there are a number of memory locations that serve more than one purpose depending on other factors (the display mode, the number of backgrounds in use, etc).

## tinygo-org/tinygo-gba

The remaining pieces would go into the new tinygo-gba repository.  This would primarily be pieces that concern higher-level GBA concerns that are not intrinsically part of the build step or code that is used by most ROMs.  For example:

- Tools and Examples that are very GBA-specific:
  - Scripts and example Makefiles to build a GBA cartridge
  - A Dockerfile for developing for GBA on windows
  - Tools for preparing assets to embed in a ROM
- Tools and scripts that may be emulator-specific or pertain mostly to developers of the gba drivers:
  - A test suite and scripts to build and view GBA ROMs to verify functionality

aykevl, 2019-12-16:
Sounds good! Some comments below.

> Flash / memory layout (e.g. specifying values to be stored in .rodata on flash cart)

Do you mean game assets like images/sprites?
I think this should follow the Go proposal here, once it is finished: https://github.com/golang/go/issues/35950

> A Dockerfile for developing for GBA on windows

Now that TinyGo supports Windows directly, I hope this isn't needed (or is supported in the regular Docker container).

> Tools for preparing assets to embed in a ROM

There may be a broader need for something like this. For example, the initial flash contents of external flash chips on many boards. It would be great to unify support for that.

> A test suite and scripts to build and view GBA ROMs to verify functionality

Something more than `tinygo run -target=gba` does? (That is, run any given GBA program in an emulator)

KatamariJr, 2020-07-09:
Is this still happening? I just excitedly went to go look in the drivers repo but there isnt any gba stuff there.

iamjackg, 2022-01-31:
Ah yeah, it's a bit sad to see this abandoned, but I understand that people's priorities shift. I found this repo: https://github.com/tinygo-org/tinygo-gba Is this where all this work was supposed to be moved to?

deadprogram, 2023-03-05:
I did not have any time to work on it further, *however* I did obtain the needed equipment to do so 😸 

[image]

I'd love to help out here, so I will probably start by following your article @scraly and see what happens.

scraly, 2023-03-05:
Thanks that's a good news 🙂
I really want to see some improvements in the GBA support

deadprogram, 2023-03-05:
Just added https://github.com/tinygo-org/tinygo/pull/3520 which should help get other GBA features implemented.