tinygo-org/tinygo #798: function is already defined:internal/bytealg.CountString

ellemlabs, 2019-12-21:
When trying to compile this code

```
package main

import (
	"strings"
)

func main() {
	attribs := strings.Split("asdf", "\n")
	println(attribs[0])
}
```

I get: `../../../../usr/local/go/src/internal/bytealg/count_generic.go:19:6: function is already defined:internal/bytealg.CountString`

aykevl, 2019-12-21:
It works for me:

```
$ tinygo run tmp/issue798.go
asdf

$ tinygo version
tinygo version 0.11.0-dev linux/amd64 (using go version go1.13)
```

Can you show the exact command you're using and your `tinygo version`?

ellemlabs, 2019-12-21:
I ran it on two different versions, bot gave the same error:

`tinygo version 0.11.0-dev linux/amd64 (using go version go1.12.10)`

and

`tinygo version 0.10.0-dev linux/amd64 (using go version go1.12.10)`

...Perhaps its my Go version (especially seeing how the source file at issue is in /usr/local/go). I'll try to upgrade that.

ellemlabs, 2019-12-21:
I forgot to put the compile command (`tinygo build -target=wasm`)... I'm building this as a wasm module.

This may be another issue entirely, but when I upgraded to go.13.5, I cannot compile anything at any target: 

```
$ tinygo run main.go
# unicode/utf8
../../../usr/local/go/src/unicode/utf8/utf8.go:28:8: expected ';', found b00000000 (and 10 more errors)
```

Looks like tinygo does not yet handle binary representations (or at least the version I have). But again, probably another issue.

aykevl, 2019-12-21:
> This may be another issue entirely, but when I upgraded to go.13.5, I cannot compile anything at any target:

It means you need to recompile TinyGo: there was a language change in Go 1.13 that is not included in TinyGo compiled with Go 1.12.

aykevl, 2019-12-21:
This issue _should_ be fixed with the following PR: https://github.com/tinygo-org/tinygo/pull/707

Tried reproducing it again with `tinygo run -target=wasm ./tmp/issue798.go` (issue798.go contains the code posted above) and still can't reproduce it even with Go 1.12:

    tinygo version 0.11.0-dev linux/amd64 (using go version go1.12)

ellemlabs, 2019-12-21:
Just needed to rebuild tinygo and update go to 1.13.5.

It compiles now.

closed 2019-12-21 by ellemlabs