tinygo-org/tinygo #805: crypto/sha256 causes $memcpy call (and thus breaks the WASM module)

ellemlabs, 2019-12-23:
I'm essentially re-opening #728 

The following code will compile, but the resulting wasm will be unusable as LLVM seems to inject a "$memcpy" call into the wasm code, which is undefined. Thus the WASM cannot be successfully linked in the browser.

It goes back to the use of "crypto/sha256"... and can be fixed with this workaround: https://github.com/tinygo-org/tinygo/issues/728#issuecomment-554686962

```
// main.go:
package main
import (
	"crypto/sha256"
	"fmt"
)
func main() {
	h := sha256.New()
	h.Write([]byte("abc"))

       // This line prevents webassembly from linking properly and will cause the error
       // LinkError: WebAssembly.instantiate(): Import #... module="env" function="memcpy" error: function import requires a callable
	b := h.Sum(nil) 

	fmt.Printf("%v", b) // this line just makes sure the compiler doesn't cut any code.
}
```

Terminal:

```
$ tinygo version
tinygo version 0.11.0-dev linux/amd64 (using go version go1.13.5)
$ tinygo build -target=wasm -o main.wasm main.go
$ wasm2wat main.wasm | grep -n "\$memcpy"
42:  (import "env" "memcpy" (func $memcpy (type 1)))
5711:    call $memcpy
```

deadprogram, 2019-12-30:
#820 should correct this, which has now been merged into `dev` 

@ellemlabs can you please verify this corrects your issue?

ellemlabs, 2019-12-31:
@deadprogram 

That seems to have fixed it. memcpy seems to now be defined. Which, in turn prevents the browser from complaining about the undefined calls.

```
$ wasm2wat main.wasm | grep "\$memcpy"
(func $memcpy (type 2) (param i32 i32 i32) (result i32)
                             call $memcpy
 (export "memcpy" (func $memcpy))
```

closed 2019-12-31 by ellemlabs