tinygo-org/tinygo #816: Setting up with Parcel

talentlessguy, 2019-12-27:
Related #410 

I tried to set Parcel up with `tinygo` but I faced the problem I'm not sure how to solve. Parcel has its own module resolution and interop with Node, so even in browser there is still a `process` object,

Which breaks `wasm_exec`:

```
wasm_exec.js:383 Uncaught TypeError: Cannot read property 'write' of undefined
    at wasm_exec.js:383
    at Object.parcelRequire.src/js/wasm_exec.js.text-encoder (wasm_exec.js:9)
    at newRequire (wasm_exec.085e43ef.js:47)
    at wasm_exec.085e43ef.js:81
    at wasm_exec.085e43ef.js:120
```

here's my current `wasm_exec.js` file:

```js
import { TextDecoder, TextEncoder } from 'text-encoder'
import 'regenerator-runtime/runtime'
;(() => {
  const isNodeJS = typeof process !== 'undefined'
  if (isNodeJS) {
    global.require = require
    global.fs = require('fs')

    const nodeCrypto = require('crypto')
    global.crypto = {
      getRandomValues(b) {
        nodeCrypto.randomFillSync(b)
      }
    }

    global.performance = {
      now() {
        const [sec, nsec] = process.hrtime()
        return sec * 1000 + nsec / 1000000
      }
    }

    const util = require('util')
    global.TextEncoder = util.TextEncoder
    global.TextDecoder = util.TextDecoder
  } else {
    if (typeof window !== 'undefined') {
      window.global = window
    } else if (typeof self !== 'undefined') {
      self.global = self
    } else {
      throw new Error('cannot export Go (neither window nor self is defined)')
    }

// ...
```

Is there any way to make it work? maybe properly resolving `process`?

aykevl, 2019-12-30:
I don't really know - I don't know a lot about modern JS and especially not about NodeJS. The wasm_exec.js file was originally copied from the Go project but modified quite a bit to make it work with TinyGo.

talentlessguy, 2019-12-30:
@aykevl ok, I'll try to play with `wasm_exec` to make it work with parcel by myself

Should I close the issue or keep it open because it doesn't work with Parcel?

deadprogram, 2024-05-05:
Closing due to age, please reopen if needed. Thanks!

closed 2024-05-05 by deadprogram