tinygo-org/tinygo #843: ld.lld-9: error: undefined symbol: _stack_top

conejoninja, 2020-01-12:
I'm using the *interrupts-gba* branch (i know it's WIP) , and I got some errors while compiling an example of tinyfont.WriteLine, but tinyfont.DrawChar works fine 

This works fine
```go
tinyfont.DrawChar(&display, &fonts.Bold24pt7b, 68, 60, byte('i'), color.RGBA{150, 150, 150, 255})
```

This produces different errors
```go
tinyfont.WriteLine(&display, &tinyfont.Picopixel, 36, 60, []byte("TinyGo"), color.RGBA{150, 150, 150, 255})
```

Seems to fail at this lines :
https://github.com/conejoninja/tinyfont/blob/master/tinyfont.go#L75 (it compiles if I manually set w,h)
https://github.com/conejoninja/tinyfont/blob/master/tinyfont.go#L46-L52 (it compiles if I comment those lines, but oviously nothing shows on the screen). Oddly enough, those lines are executed fine when using tinyfont.DrawChar

gc=conservative (default) produces the error of the title
```
➜ tinygo build -o main.gba -target=gameboy-advance -gc=conservative github.com/conejoninja/gbatest
ld.lld-9: warning: lld uses blx instruction, no object with architecture supporting feature detected
ld.lld-9: error: undefined symbol: _stack_top
>>> referenced by gc_conservative.go:0 (../../tinygo-org/tinygo/src/runtime/gc_conservative.go:0)
>>>               /tmp/tinygo867851047/main.o:(runtime.alloc)
error: failed to link /tmp/tinygo867851047/main: exit status 1
```

gc=none gives and error
```
➜ tinygo build -o main.gba -target=gameboy-advance -gc=none github.com/conejoninja/gbatest
Global is external, but doesn't have external or weak linkage!
i8* ()* @runtime.alloc
error: verification failure after LLVM optimization passes
```
gc=leaking compiles and works fine
```
➜ tinygo build -o main.gba -target=gameboy-advance -gc=leaking github.com/conejoninja/gbatest
ld.lld-9: warning: lld uses blx instruction, no object with architecture supporting feature detected
```

Code of the example : https://gist.github.com/conejoninja/6888bf7da5b084dd01ea7890bf9150cb

aykevl, 2020-01-12:
Should be fixed in the interrupts-gba-2 branch, which is based on the RISC-V work. The API is compatible but the produced code is slightly more efficient (20 fewer bytes code size, 114 fewer bytes RAM).

conejoninja, 2020-01-13:
Tested and **working**! thanks!

Should we close this issue? or wait until it's merged?

Also, I got a bunch of different issues with random bits of code, mostly math operations, for example 
```go
myvar = myvar%30 // myvar is uint8
```

results in 

```
ld.lld-9: error: undefined symbol: __umodsi3
```

should I report them too?

aykevl, 2020-01-13:
This issue is really for a specific (WIP) branch, not really any merged code (master or dev) so I'm closing this. However, feel free to comment here with more issues as they arrive.

The "undefined symbol" error is fixed with this PR: https://github.com/tinygo-org/tinygo/pull/846

closed 2020-01-13 by aykevl