tinygo-org/tinygo #844: Unsupported architecture when compiling Go with C dependencies to WASM

loredanacirstea, 2020-01-12:
MacOS 10.15.2
tinygo version 0.11.0 darwin/amd64 (using go version go1.13.6)

I am trying to compile `rqlite` to wasm, but `rqlite` has a C dependency (`sqlite`). I know tinygo has some `cgo` support, but I am not sure what to make of the errors: 

```
$ CC=gcc CPATH="/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/" tinygo build -o wasm.wasm -target wasm ./rqlited/main.go

# github.com/mattn/go-sqlite3
src/github.com/mattn/go-sqlite3/sqlite3.go:25:6: not implemented: build constraints in #cgo line
src/github.com/mattn/go-sqlite3/sqlite3_other.go:12:6: not implemented: build constraints in #cgo line
src/github.com/mattn/go-sqlite3/sqlite3_other.go:13:6: not implemented: build constraints in #cgo line
src/github.com/mattn/go-sqlite3/sqlite3_other.go:14:6: not implemented: build constraints in #cgo line
src/github.com/mattn/go-sqlite3/sqlite3_other.go:15:6: not implemented: build constraints in #cgo line
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:807:2: error: Unsupported architecture
src/github.com/mattn/go-sqlite3/backup.go:14:10: note: in file included from src/github.com/mattn/go-sqlite3/backup.go!cgo.c:7:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:62:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:62:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/machine/_types.h:34:2: error: architecture not supported
src/github.com/mattn/go-sqlite3/backup.go:14:10: note: in file included from src/github.com/mattn/go-sqlite3/backup.go!cgo.c:7:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:64:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:64:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/_types.h:27:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/_types.h:27:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:33:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:33:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:55:9: error: unknown type name '__int64_t'
src/github.com/mattn/go-sqlite3/backup.go:14:10: note: in file included from src/github.com/mattn/go-sqlite3/backup.go!cgo.c:7:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:64:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:64:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/_types.h:27:10: note: in file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/_types.h:27:
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:56:9: error: unknown type name '__int32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:57:9: error: unknown type name '__int32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:60:9: error: unknown type name '__uint32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:61:9: error: unknown type name '__uint32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:62:9: error: unknown type name '__uint64_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:68:9: error: unknown type name '__darwin_natural_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:70:9: error: unknown type name '__uint16_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:71:9: error: unknown type name '__int64_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:72:9: error: unknown type name '__int32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:73:9: error: unknown type name '__uint32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:74:9: error: unknown type name '__int32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:75:9: error: unknown type name '__uint32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types.h:76:9: error: unknown type name '__uint32_t'
../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/machine/signal.h:34:2: error: architecture not supported

.... another 230 lines of errors...
fatal: too many errors emitted, stopping now
```

aykevl, 2020-01-12:
There are multiple things that are not yet implemented in TinyGo to get go-sqlite3 to compile. I'm aware of the following:

  * Build constraints in the `#cgo` line (as the error you reported indicates). This is for a line like the following:
    
        #cgo linux,!android CFLAGS: -DHAVE_PREAD64=1 -DHAVE_PWRITE64=1
    
  * Using a libc on WebAssembly, probably wasi-libc.
  * You haven't seen it yet, but including database/sql will result in several compile errors:
    ```
    # database/sql
    ../../../../../../usr/local/go/src/database/sql/sql.go:590:7: Locker not declared by package sync
    ../../../../../../usr/local/go/src/database/sql/sql.go:3159:7: Locker not declared by package sync
    ../../../../../../usr/local/go/src/database/sql/sql.go:3181:23: Locker not declared by package sync
    ../../../../../../usr/local/go/src/database/sql/sql.go:3177:29: Stack not declared by package runtime
    ../../../../../../usr/local/go/src/database/sql/sql.go:3170:5: dr.Lock undefined (type driverResult has no field or method Lock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:3171:11: dr.Unlock undefined (type driverResult has no field or method Unlock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:3164:5: dr.Lock undefined (type driverResult has no field or method Lock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:3165:11: dr.Unlock undefined (type driverResult has no field or method Unlock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:2744:22: rs.closemu.RLocker undefined (type sync.RWMutex has no field or method RLocker)
    ../../../../../../usr/local/go/src/database/sql/sql.go:2598:5: ds.Lock undefined (type *driverStmt has no field or method Lock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:2599:11: ds.Unlock undefined (type *driverStmt has no field or method Unlock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:2427:5: ds.Lock undefined (type *driverStmt has no field or method Lock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:2428:11: ds.Unlock undefined (type *driverStmt has no field or method Unlock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:599:5: ds.Lock undefined (type *driverStmt has no field or method Lock)
    ../../../../../../usr/local/go/src/database/sql/sql.go:600:11: ds.Unlock undefined (type *driverStmt has no field or method Unlock)
    ../../../../../../usr/local/go/src/database/sql/convert.go:411:41: sv.Type().ConvertibleTo undefined (type reflect.Type has no field or method ConvertibleTo)
    ../../../../../../usr/local/go/src/database/sql/convert.go:412:13: sv.Convert undefined (type reflect.Value has no field or method Convert)
    ```

aykevl, 2020-01-12:
Take a look at #845, which should fix the "unsupported architecture" error.

aykevl, 2020-01-12:
Also,

> ```
> $ CC=gcc CPATH="/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/" tinygo build -o wasm.wasm -target wasm ./rqlited/main.go
> ```

The `CPATH` and `CC` environment variables should not be needed here. IIRC both are ignored.

deadprogram, 2020-04-14:
This has now been released, so closing. Thanks!

closed 2020-04-14 by deadprogram