tinygo-org/tinygo #850: Can wasm use httpclient?

cyanBone, 2020-01-16:
Can wasm use httpclient?

itstmyi, 2020-01-18:
FYR : https://github.com/vugu/vugu
One example : https://github.com/vugu/vugu/tree/master/examples/fetch-and-display

Good luck .

xaionaro, 2020-02-06:
@itstmyi :

*  The first line of the example is `// +build !wasm`
* TinyGo does not compile this example anyway:
```
../../usr/lib/go-1.11/src/encoding/asn1/marshal.go:537:47: invalid operation: v.Type() (value of type reflect.Type) has no field or method NumMethod
../../usr/lib/go-1.11/src/encoding/asn1/marshal.go:549:14: DeepEqual not declared by package reflect
../../usr/lib/go-1.11/src/encoding/asn1/marshal.go:558:14: DeepEqual not declared by package reflect
../../usr/lib/go-1.11/src/encoding/asn1/marshal.go:479:93: invalid operation: t.Field(startingField).Tag (value of type string) has no field or method Get
../../usr/lib/go-1.11/src/encoding/asn1/marshal.go:483:103: invalid operation: t.Field(i + startingField).Tag (value of type string) has no field or method Get
../../usr/lib/go-1.11/src/encoding/asn1/common.go:174:26: invalid operation: t (variable of type reflect.Type) has no field or method Name
../../usr/lib/go-1.11/src/encoding/asn1/asn1.go:658:80: invalid operation: ifaceType (variable of type reflect.Type) has no field or method NumMethod
../../usr/lib/go-1.11/src/encoding/asn1/asn1.go:810:115: invalid operation: fieldType (variable of type reflect.Type) has no field or method Name
../../usr/lib/go-1.11/src/encoding/asn1/asn1.go:831:12: Copy not declared by package reflect
../../usr/lib/go-1.11/src/encoding/asn1/asn1.go:919:104: invalid operation: field.Tag (variable of type string) has no field or method Get
../../usr/lib/go-1.11/src/encoding/asn1/asn1.go:932:12: Copy not declared by package reflect
```

bradleypeabody, 2020-05-12:
I'm considering making a subset of the `net/http` package intended for client use in Vugu.  The idea would be to make things like `http.Get` work the same way they do with Go's wasm implementation of that package.

Most of the things that don't compile in net/http are not useful under wasm anyway (in the browser I mean) - you cannot create raw sockets to do the various low-level things that are failing to compile.  (i.e. I'm guessing that simply omitting via build tag whatever code is calling that asn1 stuff above would probably be an effective solution for that error for wasm, at least for now.). I do understand that the situation is different on microcontrollers, where it would be really cool to have a working http client and server for environments where it's possible.

When I put this alternate http package together for Vugu I'll also create an issue here to discuss if there would be any use in having this as part of TinyGo's stdlib.  (A similar question exists for [vjson](https://github.com/vugu/vjson), which is a subset of encoding/json that uses type switches and so does not require reflect.New, etc.)

bgould, 2020-05-12:
@bradleypeabody what you are saying is very doable... we have this for microcontrollers which was intended to be a bridge towards having a workable `net` package on platforms with no OS: https://github.com/tinygo-org/drivers/tree/master/net - currently it allows for having a unified API whether you are using the `espat` or `wifinina` packages, though I think it could definitely use some work.  I'm not sure if that would be useful for wasm but just pointing it out because it sort of proves out a concept.

As far as http goes, I've been able to modify and use https://github.com/gorilla/http on top of the drivers/net package and it works well enough, though it is not exactly a `net/http`-like API.  I haven't looked at it in a bit so I'm not sure what shape its in but I could try to get it ready to share soon.  I also had this all working in combination with https://github.com/valyala/fastjson (but it was kind of flaky, might work better in wasm).

All this is to say that I think you idea is very feasible and I look forward to seeing the result.

aykevl, 2020-05-12:
I think that could work in the short term. However, in the longer term I want to make it possible to compile the standard library net/http package. I believe it has some support for running under WebAssembly in a browser.

bradleypeabody, 2020-05-12:
Cool, when I dive in on this I will look at that possibility too - perhaps just a few small build tag changes would allow net/http to compile (although it would require forking that package, but it might be worth it and if the changes are simple it should be easy to maintain, possibly even scriptable).

davecheney, 2026-10-24:
Consolidating browser HTTP client support into https://github.com/tinygo-org/tinygo/issues/4420, which describes the requirement for standard net/http calls backed by browser networking in more detail. This does not mean browser HTTP support is implemented.

This minimal example demonstrates the requested behavior. It has not been run:

```go
package main

import (
	"io"
	"net/http"
)

func main() {
	resp, err := http.Get("/example.txt")
	if err != nil {
		panic(err)
	}
	defer resp.Body.Close()

	body, err := io.ReadAll(resp.Body)
	if err != nil {
		panic(err)
	}
	println(string(body))
}
```

This is intended for a browser host serving example.txt from the same origin.

davecheney, 2026-10-24:
Duplicate of #4420

closed 2026-10-24 by davecheney