tinygo-org/tinygo #851: "invalid file format" while flashing code to Arduino Uno via avrdude on Win10

alankrantas, 2020-01-16:
I've actually found a way around it, but it would be nice if I can do this in a single step.

My setup:

Windows 10 version 1909
tinygo version 0.11.0 windows/amd64 (using go version go1.13.6)
PATH to Go, TinyGo, Git and avr-gcc/avrdude (using [avr-gcc-9.2.0-x64-mingw](https://blog.zakkemble.net/avr-gcc-builds/)) bins added
Target is Arduino Uno.

If I use commands like this in the command prompt

<b>tinygo flash -target arduino -port COM15 examples/blinky1</b>

I would get the following messages

```
avrdude: invalid file format '\Users\xxx\AppData\Local\Temp\tinygo220029627\main.hex' in update specifier
avrdude: error parsing update operation 'flash:w:C:\Users\xxx\AppData\Local\Temp\tinygo220029627\main.hex'
error: failed to flash C:\Users\xxx\AppData\Local\Temp\tinygo220029627\main.hex: exit status 1
```

Then I tried to flash the main.hex to my Uno using avrdude. But I would get error messages that the generated .hex in the AppData\Local\Temp directory does not exist (and it did; already deleted by tinygo?)

However, the following commands works:

<b>tinygo build -target arduino -o "C:\tinygo\blinky1.hex" examples/blinky1</b>

<b>avrdude -p atmega328p -c arduino -P COM15 -b 115200 -D -U flash:w:"C:\tinygo\blinky1.hex":a</b>

And the flashing was successful:

```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\tinygo\blinky1.hex"
avrdude: input file C:\tinygo\blinky1.hex auto detected as Intel Hex
avrdude: writing flash (490 bytes):

Writing | ################################################## | 100% 0.13s

avrdude: 490 bytes of flash written
avrdude: verifying flash memory against C:\tinygo\blinky1.hex:
avrdude: load data flash data from input file C:\tinygo\blinky1.hex:
avrdude: input file C:\tinygo\blinky1.hex auto detected as Intel Hex
avrdude: input file C:\tinygo\blinky1.hex contains 490 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.11s

avrdude: verifying ...
avrdude: 490 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

This method also works on several other example codes.

deadprogram, 2020-01-18:
The tinygo flash command in Windows was revised last release to handle Arduino Nano33 and all of the UF2 bootloader ARM boards from Adafruit, however I did not get around to doing the same for AVR boards like the Uno yet.

A PR that added this would be most gratefully merged. Otherwise I will try to work on it for next release.

alankrantas, 2020-01-18:
Sorry, it's beyond my capability to do a PR. I'm interested in writing teaching materials for TinyGo some time later though. We don't have many choices in east Asia; China-made Uno (and ESP8266 boards if they are added in the future) are the cheapest and most common ones for teaching purposes.

Also **tinygo flash** works with my micro:bit under Win10.

deadprogram, 2020-01-20:
Good to know.

Yes the support for running avrdude on Windows automatically via "tinygo flash" is not yet implemented, but the ARM-based platforms like Microbit, the Adafruit board, etc. should all work.

We will see what we can do to get the avrdude support in soon.

aykevl, 2020-01-20:
@alankrantas can you try the patch in https://github.com/tinygo-org/tinygo/pull/857? You can simply edit the targets/arduino.json file of your TinyGo installation.

> I'm interested in writing teaching materials for TinyGo some time later though.

Cool!

> We don't have many choices in east Asia; China-made Uno (and ESP8266 boards if they are added in the future) are the cheapest and most common ones for teaching purposes.

Yeah I really wish Espressif was a bit more proactive with getting it merged in LLVM. But without upstream support, supporting the ESP8266 is going to be difficult.

Be aware that the AVR support (such as in the Uno) is rather limited, and you will hit compiler bugs in any significant amount of code. So I'm not sure if it's the best for teaching.
Have you looked into the so-called bluepill board? I don't know what is easily available in Asia but it's all over the place on AliExpress (sold as stm32 development board for around $2). The board is a bit more hassle to work with (you need a separate programmer, also for around $2) but the compiler support for that is stable, unlike AVR.
You could also take a look at a board that is sold as "WeMos d1 samd21". It's not yet directly supported in TinyGo but adding it should be trivial. It is a bit more expensive, but the samd21 has excellent support in TinyGo.

alankrantas, 2020-01-21:
Patch #857 works. Thanks! Only one character difference.

Please consider to add a target for Arduino Nano too (add "-b 57600" in the flash command; that works for me too).

I'll see if I can find cheap STM32 or SAMD21 boards. But generally only the lower-level Arduino and ESP family can be easily bought in both online and physical stores. My company mostly produce consumer kits for Arduino Nano and Wemos D1 mini, on Arduino C or MicroPython. These boards are bought in bulk on Taobao.com which can be even cheaper. The D1 mini are like less than $2 or $3 USD each.

I'm going to make a proposal based on Uno (may be rejected though) and maybe I'll write something for micro:bit on my own too. In some way it can also serves as an interactive Golang learning kit, which may be real popular later.

closed 2020-01-21 by alankrantas

aykevl, 2020-01-21:
> Please consider to add a target for Arduino Nano too (add "-b 57600" in the flash command; that works for me too).

See #859

alankrantas, 2020-01-30:
Just some follow-up.

My proposal didn't get through, so it'll be some sort of personal project, mostly will focus on Arduino Uno and micro:bit (when its ADC and PWM are supported).

There's actually quite a few people selling those "bluepill board" and various STM32 boards - but they don't look so user friendly. I decided to order an Adafruit Metro M4 first since I'd like to try CircuitPython and Maker MakeCode as well. Probably will try out an SAMD21-based Arduino later.

aykevl, 2020-01-30:
Okay, that's sad to hear.

Yes the M4 (samd51) is almost as well supported as the M0 (samd21) and will likely only improve in the future.

PR #857 (2020-01-21): arduino: make avrdude command line compatible with Windows
On Windows, it is common that there is a colon in the path. `avrdude` will treat that as a separator and everything behind it as the file format specifier instead of defaulting to Intel hex format.

By explicitly specifying the Intel hex format (with `:i`), this issue should be fixed.

fixes #851 (hopefully)