tinygo-org/tinygo #854: undeclared name: C.CString

gertcuykens, 2020-01-19:
I will create repo later to have full example. Here is just `main.go` for now that works in normal go but not in tinygo.

```sh
% tinygo build -o a.out
# c
main.go:13:7: undeclared name: C.CString
main.go:15:12: cannot initialize 2 variables with 1 values
main.go:17:11: undeclared name: C.hello
main.go:19:11: undeclared name: C.hello3
```

```go
package main

// #cgo CFLAGS: -D BUILD=5
// #include <stdlib.h>
// #include "file.h"
// #include "hello.h"
// #include "cgo_hello.h"
import "C"
import "unsafe"
import "fmt"

func main() {
	s := C.CString("unexist.txt")
	defer C.free(unsafe.Pointer(s))
	n, err := C.file(s)
	fmt.Println(n, err)
	n, err = C.hello()
	fmt.Println(n, err)
	n, err = C.hello3()
	fmt.Println(n, err)
}

//export hello2
func hello2() int{
	fmt.Println("hello form go")
	return 0
}
```
PS maybe introduce a github cgo label to tag all cgo issues. Thanks

aykevl, 2020-01-19:
Yes these helper functions have not been implemented yet. They're blocked on libc support, which I'm working on in #845 and #849.

deadprogram, 2021-11-06:
I am pretty sure this feature now exists ever since libc support was merged. Now closing, please reopen if needed. Thanks!

closed 2021-11-06 by deadprogram

aykevl, 2021-11-11:
No, this feature has not yet been implemented. Libc support is now in place (which is a prerequisite), but this feature still doesn't exist in TinyGo.