tinygo-org/tinygo #855: Set PWM frequency

alankrantas, 2020-01-20:
Can you add a method to set PWM's frequency in the near future? It would be useful for controlling passive piezo buzzers and servos. (I'm having trouble using the buzzer driver in repo tinygo-org/drivers on my Arduino Uno.)

Thanks.

aykevl, 2020-01-30:
I don't think this is as simple as it seems. Usually chips can control multiple PWM channels (where one channel is usually mapped to one pin) with a single PWM peripheral that runs on a given clock. This means that you can only change the frequency for multiple channels/pins at once.

Ideally we would need to look at a few chips from a few different vendors to see how the hardware usually works, to make sure such an API would map cleanly to the underlying hardware on as many chips as possible.

gwtnz, 2020-02-22:
Piezos/buzzers and servos (and LEDs and motors) likely need different combos of duty cycle and frequency (which will affect the resolution of the output).

e.g. an LED / piezo / buzzer can probably do a duty cycle of 0%-100% at whatever frequency you want, e.g. 1000Hz frequency, 10% duty cycle => 100ms on, 900ms off, and the resolution is probably fine in 1% increments (10ms)

a consumer servo has very specific needs: 40-50Hz pulse rate (20-25ms), resolution ~180 degrees ( 0-255 is probably fine for most applications) which has to span a range between about 1-2ms => the clock cycles need to be 1ms/256 = ~4ms => ~250kHz

But as @aykevl says, running servos and LEDs at the same time on the same timer hardware is not trivial. I'm planning on implementing some attempts at it on the `stm32f4` discovery, as I have some other ideas too about expanding the number and choice of pins beyond those that are connected in hardware to the timer.

hbroer, 2020-09-05:
Hi, I also need to set the frequency on a atsamd21 to be above 20khz. I have some code for C++ but I have no idea how to translate that to Golang. I even do not understand whats going on in C++, but it works the last time I played around with that. Code can be found here somewhere in this thread: [forum.arduino.cc](https://forum.arduino.cc/index.php?topic=346731.0)

It would be perfect if I could set the frequency globally with `machine.InitPWM(20000)` or with `machine.SetPWMFrequency(20000)`. If that's not possible, how can I learn how to configure it myself with TinyGo?

aykevl, 2021-05-13:
This is now implemented in the latest release, see https://github.com/tinygo-org/tinygo/pull/1121.

closed 2021-05-13 by aykevl