tinygo-org/tinygo #864: Wasm: how to set __heap_base ?

vshymanskyy, 2020-01-22:
I run into this problem while running a wasm file on ESP32 and other devices (yeagh..).
I'm using [Wasm3](https://github.com/wasm3/wasm3) interpreter, and created this example:

https://github.com/wasm3/wasm3-arduino/tree/master/wasm_apps/tinygo

Now I'm trying to use the Linear Memory, but many embedded devices cannot afford to allocate even a single Wasm page (64KiB), so we allocate as much as we can... And it works with AssemblyScript, Rust for example.
In TinyGo, I get `__heap_base = 65536` and can't find a way to lower it.

[image]

With this picture in mind I tried:
```sh
tinygo  build -target wasm                \
        -panic trap -wasm-abi generic     \
        -heap-size 4096                  \
        -ldflags="-z,stack-size=2048" \
        -o app.wasm app.go
```
Still getting `__heap_base = 65536`.

Would appreciate any help. Thanks!

aykevl, 2020-01-30:
I believe this is a linker limitation. Ideally `__heap_base` would be put right after the end of the static data (stack and `.data`) but IIRC I couldn't find a way to do that. If you have ideas how to fix that, that would be great (such as a special symbol for the end of `.data`).

However, there is one way you can influence the heap size and that is with the `-heap-size` flag. Unfortunately, IIRC it is not possible to set it to something other than a multiple of 64K otherwise `wasm-ld` will complain.

I would like to see TinyGo supported by this wasm runtime so I hope this can be fixed somehow.

vshymanskyy, 2020-01-31:
@aykevl  I think I achieved (at least patially) what I wanted with `-ldflags="-z stack-size=2048 --max-memory=65536"` flag.
Yup.. the problem was hidden in the `,` character ;)

closed 2020-01-31 by vshymanskyy