tinygo-org/tinygo #865: Package organization suggestion

FPabl0, 2020-01-22:
I really love this project. After looking at the project organization, I think it should be better to isolate the code from the different targets in different packages that can be downloaded from the users, instead of forcing them to install all of them. Also, in this way, more people can generate their own HAL for specific targets.

deadprogram, 2020-01-28:
Hi @FPabl0 thanks for the suggestion.

It certainly might be possible to do further organization, however moving each of the supported platforms into a separate package is not really something that we probably desire. 

I think most of us really prefer the monorepo style. Among other benefits, it improves discoverability IMO, instead of performing a "scavenger hunt" across the internet looking for clues/packages. Other languages have taken the many repos approach, and I have found it very difficult to search for what I need.

There is not any install difference when using TinyGo with different supported MCUs, unless you are referring to the extra steps needed in the case that you want to use AVR-based boards like the Arduino Uno. Unless you are referring to something else, in which case please elaborate.

Anyhow, thanks for the feedback!

aykevl, 2020-01-28:
In addition to what @deadprogram said, splitting out such packages is difficult in practice because the compiler/runtime has to know at least some details of the target, such as: startup code, sleeping for a given time, getting the current time (since reset), and writing to a debug output (such as serial). Additionally, the machine package heavily relies on chip specific features to implement a common HAL.

If you have any concrete suggestions how targets can be moved to separate packages, I'd love to hear them.

If it's about binary size, then I think this can be improved. For example, the generated Go and assembly files in src/device/* take up a lot of space and will only continue to grow. It may be useful to eventually move those into separate vendor-specific packages that are automatically downloaded (and updated, after a release) by TinyGo. I think we should use Go modules for them, once they become supported in TinyGo. But even with such an organization, the release would still be one whole, just packaged differently.

fpabl0, 2020-02-23:
Hi @deadprogram @aykevl , thanks for your replies. Yes, monorepo style is good preventing useless packages, but my purpose is not let the users creates their own package in different places, it is to move the different microcontroller families into different repos but inside this official github account, it is just for better readability and can be useful because there can be teams (with some experience in a specific family) in charge of for example avr microcontrollers, other for stm32 microcontrollers, other for nrf, etc. In that way, I think TinyGo will have a better support. For example, I would like to support stm32g0 series, but I find a little hard to do it, because all the microcontroller families are messy in the same place.

A microcontroller family repo may have this structure:
`
stm32-repo/
    \_ machine/
    \_ runtime/
    \_ boards/
    \_ targets/
`

It is just for development, so maybe the TinyGo CLI should have a command to put the repo of a family inside the machine/runtime package, to let it work just as now.

Keep in that way, this is a great project!!
Sorry for my english.
Greetings from Ecuador.

deadprogram, 2020-02-24:
Thanks for the suggestions, we will keep them in mind for the future.

Now closing, please reopen if needed.

closed 2020-02-24 by deadprogram

aykevl, 2020-02-26:
Ah, that makes sense. Yes I think that is a good idea and I think it's a good idea to move in that direction eventually. But with this the details are very important.

One practical way this could be done is by moving all the peripheral implementations to subpackages of the machine packages. So for example you'd have machine/nrf, machine/stm32, machine/sam, etc. (but not sure how to avoid namespace conflicts with device/nrf etc). In such a scenario each board/chip would still need such files in the bare machine package (to define pin mappings and peripherals) but it should be doable. That would limit the code to one directory and one board file in another directory, without having to wade through all the HAL for _all_ supported boards.

fpabl0, 2020-03-07:
Hello, I have been working in this suggestion. I have tested it with a nucleo-g070rb and works fine at least a blinky with go routines. Basically, in the runtime package we could have just one file per microcontroller family (runtime_avr.go, runtime_stm32.go). This file could be written like this:

```
// +build stm32

package runtime

import "github.com/tinygo-org/tinygo-stm32/runtime"

type timeUnit = runtime.TimeUnit

const (
	asyncScheduler = runtime.AsyncScheduler
	tickMicros     = runtime.TickMicros
)

//go:export Reset_Handler
func main() {
	preinit()
	initAll()
	callMain()
	abort()
}

func init() {
	runtime.DeviceInit()
}

func putchar(c byte) {
	runtime.Putchar(c)
}

func sleepTicks(d timeUnit) {
	runtime.SleepTicks(d)
}

func ticks() timeUnit {
	return runtime.Ticks()
}
```

The import package "github.com/tinygo-org/tinygo-stm32/runtime" is where all the logic of the family is developed. This an example that covers just the stm32g070rb:

[https://github.com/fpabl0/tinygo-stm32](url)

And this is a blinky for stm32g0:

https://github.com/fpabl0/tinygo-blinky-test